realthunder/FreeCAD_assembly3 #1: How to test this workbench?

andy-kliman, 2017-10-11:


realthunder, 2017-10-11:
Thanks for the interest. But, it's not ready yet. I'll post up instructions once it's ready. It only works on my heavily patched FCAD, so you'll need to be able to compile FCAD yourself in order to test it.

andy-kliman, 2017-10-11:
Where can I find the sources of the patched FCAD?

I tried to run the workbench by copying the contents to the MOD folder, but the console displays the messages "No module named asm3 occurred"

realthunder, 2017-10-11:
Most likely https://github.com/realthunder/FreeCAD/tree/LinkStage3.8, but may change. This WB also serves as a test ground for my FCAD branch. At the moment, both this and my FCAD repo are under frequent changes (I am doing rebase a lot since they are not published yet). So it is best to just wait for a few weeks before getting your hands dirty.

andy-kliman, 2017-10-11:
Ok
In the future the workbench can be used with the standard FCAD build or not?

realthunder, 2017-10-11:
That's my goal, of course. I am trying to get my branch merged into upstream, but it is going to be a long process, because it has quite a few core changes. This WB is also my attempt to accelerate the process, you know, to show them that those core changes are worthy.

andy-kliman, 2017-10-11:
I wish good luck, I will follow up

luzpaz, 2017-12-05:
Now that the README is live we can ask @yorikvanhavre to put this WB in FreeCAD-Addons so it will be easy for folks to download in to this said forked version of FC that you mention.

yorikvanhavre, 2017-12-05:
I prefer to wait until @realthunder tells me it's good to go, then I add it to the FreeCAD-Addons.
BTW, Fantatic job @realthunder !!! I really like that incremental approach, building after other, earlier experiments (assembly2), that's the best way we get to build a real assembly workflow IMHO

realthunder, 2017-12-14:
A Windows pre-built image is now available at https://github.com/sgrogan/FreeCAD/releases/tag/RTA3

realthunder, 2017-12-20:
Linux AppImage is now available [here](../releases/latest)

efferre79, 2018-07-12:
@realthunder : which branch of your freecad fork are you usually using these days?

realthunder, 2018-07-12:
FreeCAD releasing branch is [LinkStage3](https://github.com/realthunder/FreeCAD/tree/LinkStage3).

timmwille, 2021-01-23:
it is time to test right? at least from the edit here:
https://www.forum.freecadweb.org/viewtopic.php?f=34&t=45349

by the way while looking for a way to get this running I found the very origins how this was mentioned to the FC developers first:
https://gitter.im/probonopd/AppImageKit?at=5a2683ef3a80a84b5bbc228c

thank you so much for all the work @realthunder and all the helping hands around! warms up the open source hardware engineers heart to see this grow

luzpaz, 2021-02-20:
We can close this ticket as well.

closed 2021-02-21 by realthunder