realthunder/FreeCAD_assembly3 #3: Symoro as backend

galou, 2017-11-30:
Thank you for tackling the improvement of the assembly workbench!

I would like to suggest Symoro as alternative backend (https://github.com/symoro/symoro). Symoro is also based on Sympy. It may be easier to implement than Sympy directly because it uses the Khalil-Kleinfinger notation for serial/tree/closed chains (https://www.researchgate.net/publication/224729906_A_new_geometric_notation_for_open_and_closed-loop_robots).

The advantage of Symoro is that we could also export the kinematics/dynamics equations as symbolic expressions in C++.

realthunder, 2017-12-02:
Thanks for the suggestion. I'll take look once got time. Meanwhile, I first need to get the Solvespace solver fully working first. My SymPy backend is already working under some simple tests. But the kinematics capability of Symoro is definitely worth looking into. Anyway, that's the whole purpose of a multi-backend architecture, to make it easy to try new things.

easyw, 2017-12-02:
@galou @realthunder 
it seems that simoro doesn't support py3

> Requirements
> 
>     python (>= 2.7,    3.* is not supported)

FreeCAD is going toward py3 compatibility...

galou, 2017-12-05:
I can't remember exactly why Python 3 is not supported but I guess this is because of one of the dependencies. I'll try to find the time to ensure/test the compatibility with Python 3.

maidenone, 2018-08-05:
@galou, did you find time? Looked through the paper and project, should not take long to port the project. I can take a go at porting it as i want to learn the notation / control theory as well as improve my py3..

galou, 2018-08-06:
I actually already ported Symoro to Python 3 but my PR (https://github.com/symoro/symoro/pull/15) hasn't been merged.
I'll contact the developer.

luzpaz, 2022-07-01:
Looks like symoro isn't hacked on anymore @galou. Please advise?

galou, 2022-09-20:
Unfortunately time is missing. I don't have the permissions to edit the repo, which also doesn't help.