realthunder/FreeCAD_assembly3 #4: FR: add angle rotation constraint

easyw, 2017-12-05:
Hi @realthunder 
thanks a lot for going into this 'A2 reloaded' WB!
A lot of new feature are going to be added to FC :smile:
May I ask you if you would consider to add also an angle rotation constraint?
I linked a reference topic at FC forum
https://forum.freecadweb.org/viewtopic.php?f=3&t=24625
and a issue raised at A2 WB
https://github.com/hamish2014/FreeCAD_assembly2/issues/145
It would be nice to have a constraint to define an angle between two objects with the same axis... the angle would define a rotation referred to the plane orthogonal to the common axis...
Maurice

realthunder, 2017-12-05:
It does have an angle constraint. I am not sure if this is what you want. This constraint is pretty sensitive to initial condition. You'll need to nudge the object a little bit to get the snap.
[image]

easyw, 2017-12-05:
Hi @realthunder 
the angle constraint I was asking for is not an angle between objects axis, but an angle fixed between the rotation around the axis...
I'm adding an image of an angle grade of freedom that I would like to have configurable... i.e. 45degrees between the two arrows
[image]

realthunder, 2017-12-06:
In that case, you just need two constraints, one axial alignment, and one angle.

[image]

In case where the angle constrained object has no planar face for alignment, simply create some auxiliary face/edge, say using a sketch, group the object and auxiliary object into a assembly, and then add it to the main assembly. You can hide the auxiliary object, and select it in the tree view to create constraint.

I will ensure better support of PartDesign body in the future, so that any datum/origin/workplane can be used for constraint. Haven't got time to test it.

easyw, 2017-12-06:
Ah ah you are cheating me :wink:
My 3D models have this option to select a face and the apply an angle plane constraint...
but it is not the case of the user case I posted above
[image]

> In case where the angle constrained object has no planar face for alignment, simply create some auxiliary face/edge, say using a sketch, group the object and auxiliary object into a assembly, and [...]

ok, that could be a workaround working with A3 (not with A2) ... but a direct angle constraint on main normal to axis plane it would be really nice / easier

realthunder, 2017-12-06:
So, instead of the lock rotation option in A2 axial constraint, you want an angle option. Am I right? If so, I think that's quite doable with SloveSpace. I'll add that in the future. 

It's actually the _lock rotation_ option that I am not sure how to implement. Maybe have a hidden property to remember the current angle and then add it as an implicit angle constraint.

easyw, 2017-12-06:
> So, instead of the lock rotation option in A2 axial constraint, you want an angle option. Am I right? If so, I think that's quite doable with SloveSpace. I'll add that in the future.

that would be great! :smile:
> It's actually the lock rotation option that I am not sure how to implement. Maybe have a hidden property to remember the current angle and then add it as an implicit angle constraint.

+1
there is already a lock rotation option in A2 ... may be this would help in find an option
[image]

realthunder, 2017-12-09:
Added LockAngle option in c2662d8cfdf079b5fac98250613c1eb30dc506b1. This option is available to 
* PlaneCoincident
* PlaneAlignment
* AxialAlignment
* MultiParallel

[image]

easyw, 2017-12-09:
great! very useful addition! 😄 
looking forward to see A3 in FC main branch
Maurice

easyw, 2017-12-09:
@gipsea you may find this interesting 😸