realthunder/FreeCAD_assembly3 #7: "SWIG_ADD_MODULE is deprecated" with Cmake>3.8

kwahoo2, 2018-01-13:
Complete log:

```
adi@adi-MS-7A34:~/kompilacje/FreeCAD_assembly3/slvs/build$ cmake -DBUILD_PYTHON=1 ..
CMake Error at cmake/GetGitCommitHash.cmake:8 (file):
  file failed to open for reading (Not a directory):

    /home/adi/kompilacje/FreeCAD_assembly3/slvs/.git/HEAD
Call Stack (most recent call first):
  cmake/GetGitCommitHash.cmake:35 (get_git_commit_hash)
  CMakeLists.txt:23 (include)

-- Using in-tree libdxfrw
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.10") 
-- Found PythonInterp: /usr/bin/python (found version "2.7.14") 
-- python version 2.7
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable exact version "2.7.14") 
CMake Deprecation Warning at /usr/share/cmake-3.9/Modules/UseSWIG.cmake:228 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  src/swig/python/CMakeLists.txt:34 (swig_add_module)

-- Configuring incomplete, errors occurred!
See also "/home/adi/kompilacje/FreeCAD_assembly3/slvs/build/CMakeFiles/CMakeOutput.log".
adi@adi-MS-7A34:~/kompilacje/FreeCAD_assembly3/slvs/build$ ls
CMakeCache.txt  CMakeFiles  exposed  extlib  src  tools
adi@adi-MS-7A34:~/kompilacje/FreeCAD_assembly3/slvs/build$ ls
CMakeCache.txt  CMakeFiles  exposed  extlib  src  tools
adi@adi-MS-7A34:~/kompilacje/FreeCAD_assembly3/slvs/build$ rm -f CMakeCache.txt 
adi@adi-MS-7A34:~/kompilacje/FreeCAD_assembly3/slvs/build$ cmake -DBUILD_PYTHON=1 ..
CMake Error at cmake/GetGitCommitHash.cmake:8 (file):
  file failed to open for reading (Not a directory):

    /home/adi/kompilacje/FreeCAD_assembly3/slvs/.git/HEAD
Call Stack (most recent call first):
  cmake/GetGitCommitHash.cmake:35 (get_git_commit_hash)
  CMakeLists.txt:23 (include)

-- The C compiler identification is GNU 7.2.0
-- The CXX compiler identification is GNU 7.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Using in-tree libdxfrw
-- Found SPACEWARE: /usr/lib/libspnav.so  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'zlib'
--   Found zlib, version 1.2.11
-- Checking for module 'libpng'
--   Found libpng, version 1.6.34
-- Checking for module 'fontconfig'
--   Found fontconfig, version 2.11.94
-- Checking for module 'json-c'
--   Found json-c, version 0.12.1
-- Checking for module 'glew'
--   Found glew, version 2.0.0
-- Checking for module 'freetype2'
--   Found freetype2, version 20.0.14
-- Checking for modules 'gtkmm-2.4;pangomm-1.4;x11'
--   Found gtkmm-2.4, version 2.24.5
--   Found pangomm-1.4, version 2.40.1
--   Found x11, version 1.6.4
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.10") 
-- Found PythonInterp: /usr/bin/python (found version "2.7.14") 
-- python version 2.7
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable exact version "2.7.14") 
CMake Deprecation Warning at /usr/share/cmake-3.9/Modules/UseSWIG.cmake:228 (message):
  SWIG_ADD_MODULE is deprecated.  Use SWIG_ADD_LIBRARY instead.
Call Stack (most recent call first):
  src/swig/python/CMakeLists.txt:34 (swig_add_module)

-- Configuring incomplete, errors occurred!
See also "/home/adi/kompilacje/FreeCAD_assembly3/slvs/build/CMakeFiles/CMakeOutput.log".

```

realthunder, 2018-01-14:
Thanks for reporting. Fixed now. Please update both asm3 and slvs submodule and try again.

kwahoo2, 2018-01-14:
Ok, it works fine now.

closed 2018-01-14 by kwahoo2