realthunder/FreeCAD_assembly3 #14: py3-problem (AttributeError, type object 'AsmCmdAddWorkplane' has no attribute 'getName')

looooo, 2018-06-30:
FreeCAD doesn't start with this workbench installed using the link-stage branch. This problem is maybe related to py3. I haven't got solvespace and pysolvespace installed, so this could be also a problem. Looking at the code it seems like the alternative solver is imported if solvespace is not available.

realthunder, 2018-07-01:
I am using a lot of metaclass functionality in python. Python3 has different syntax of setting metaclass. I used to use `future` for working in both 2 and 3, but the `future` module is not included in Windows libpack or AppImage. Do you know how to pre-package the `future` module?

realthunder, 2018-07-01:
The commit that I reverted the future and past module is 15babab61d2d6f684e31bc4cb5ef7e9eb4975dfe

looooo, 2018-07-01:
I guess there is only one solution to this problem:
__include the future module in freecad for windows__ such a shame it's not included.

looooo, 2018-07-29:
I guess this is solved with including six.

closed 2018-07-29 by looooo