realthunder/FreeCAD_assembly3 #17: FreeCAD freezes for a long period

ceremcem, 2018-07-18:
While moving and rotating parts, FreeCAD window freezes for at least 30 seconds. I'm running 0.7-AppImage on Debian 64bit. At the freezing moment, last `strace` output is as follows: 

```
stat("/tmp/.mount_FreeCADiw2fU/usr/lib/freecad-daily/Ext/freecad/asm3/mover", 0x7ffeb871dd50) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tmp/.mount_FreeCADiw2fU/usr/lib/freecad-daily/Ext/freecad/asm3/mover.x86_64-linux-gnu.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tmp/.mount_FreeCADiw2fU/usr/lib/freecad-daily/Ext/freecad/asm3/mover.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tmp/.mount_FreeCADiw2fU/usr/lib/freecad-daily/Ext/freecad/asm3/movermodule.so", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tmp/.mount_FreeCADiw2fU/usr/lib/freecad-daily/Ext/freecad/asm3/mover.py", O_RDONLY) = 13
fstat(13, {st_mode=S_IFREG|0664, st_size=14829, ...}) = 0
openat(AT_FDCWD, "/tmp/.mount_FreeCADiw2fU/usr/lib/freecad-daily/Ext/freecad/asm3/mover.pyc", O_RDONLY) = -1 ENOENT (No such file or directory)
fstat(13, {st_mode=S_IFREG|0664, st_size=14829, ...}) = 0
read(13, "import math\nfrom collections imp"..., 8192) = 8192
read(13, "   assembly = info.Parent.getLin"..., 8192) = 6637
read(13, "", 8192)                      = 0
close(13)                               = 0
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
poll([{fd=7, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=7, revents=POLLOUT}])
writev(7, [{iov_base="\1\30\v\0\366\1\200\4\26\1\200\4\0\0\0\0d\0\36\0\0\0\1\0\371\0\0\0\n \0\0"..., iov_len=152}, {iov_base=NULL, iov_len=0}, {iov_base="", iov_len=0}], 3) = 152
poll([{fd=7, events=POLLIN}], 1, -1)    = 1 ([{fd=7, revents=POLLIN}])
recvmsg(7, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\335?\22\0\0\0008\1\0\0\211\n\256\0\22\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 104
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=7, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=7, revents=POLLOUT}])
writev(7, [{iov_base="\24\0\6\0\211\n\256\0\371\1\0\0\6\0\0\0\0\0\0\0\1\0\0\0", iov_len=24}, {iov_base=NULL, iov_len=0}, {iov_base="", iov_len=0}], 3) = 24
poll([{fd=7, events=POLLIN}], 1, -1)    = 1 ([{fd=7, revents=POLLIN}])
recvmsg(7, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\336?\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 32
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=7, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=7, revents=POLLOUT}])
writev(7, [{iov_base="\17\0\2\0\211\n\256\0", iov_len=8}, {iov_base=NULL, iov_len=0}, {iov_base="", iov_len=0}], 3) = 8
poll([{fd=7, events=POLLIN}], 1, -1)    = 1 ([{fd=7, revents=POLLIN}])
recvmsg(7, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\337?\20\0\0\0008\1\0\0008\1\0\0\20\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 96
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=7, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=7, revents=POLLOUT}])
writev(7, [{iov_base="(\0\4\0\26\1\200\4\211\n\256\0\0\0\0\0", iov_len=16}, {iov_base=NULL, iov_len=0}, {iov_base="", iov_len=0}], 3) = 16
poll([{fd=7, events=POLLIN}], 1, -1)    = 1 ([{fd=7, revents=POLLIN}])
recvmsg(7, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\1\340?\0\0\0\0\26\1\200\4\0\0\23\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0", iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 32
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=7, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=7, revents=POLLOUT}])
writev(7, [{iov_base="\3\0\2\0\211\n\256\0\16\n\2\0\211\n\256\0", iov_len=16}, {iov_base=NULL, iov_len=0}, {iov_base="", iov_len=0}], 3) = 16
poll([{fd=7, events=POLLIN}], 1, -1)    = 1 ([{fd=7, revents=POLLIN}])
recvmsg(7, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\1\0\341?\3\0\0\0A\0\0\0\1\0\n\n\377\377\377\377\0\0\0\0\0\0\2\0@\0\0\0"..., iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 76
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=7, events=POLLIN|POLLOUT}], 1, -1) = 1 ([{fd=7, revents=POLLOUT}])
writev(7, [{iov_base="\22\0\30\0\26\1\200\4(\0\0\0)\0\0\0 \0\0\0\22\0\0\0\37\2\0\0\0\0\0\0"..., iov_len=120}, {iov_base=NULL, iov_len=0}, {iov_base="", iov_len=0}], 3) = 120
poll([{fd=7, events=POLLIN}], 1, -1)    = 1 ([{fd=7, revents=POLLIN}])
recvmsg(7, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\34\0\343?\26\1\200\4(\0\0\0\211\270\33\5\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., iov_len=4096}], msg_iovlen=1, msg_controllen=0, msg_flags=0}, 0) = 84
recvmsg(7, {msg_namelen=0}, 0)          = -1 EAGAIN (Resource temporarily unavailable)
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
write(4, "\1\0\0\0\0\0\0\0", 8)         = 8
```

realthunder, 2018-07-18:
Which mover did you use? Is this problem happening consistently or just occasionally?

Also, please provide more logging information by typing the following commands in Python console
```python
App.setLogLevel('asm',4)
App.setLogLevel('asm.main',4)
```

Then open the report view (View->Panels->Report View). Do the moving operation, and after it unfreezes. Right click the report view and select save log, and attach the log file here. Thanks!

ceremcem, 2018-07-19:
It freezes consistently. 

This time I even didn't need to move anything. Just opened the last saved file, clicked to rotate, then it freezed. 

I set the variables in the Python console, then tried to rotate (not move) the model and it freezed. After freeze, no additional logs are generated: 

[image]

Here is the freeze in action: 

[image]

It turns out that it's not related with moving parts, it seems to be related with a specific part in the design which has lots of details:

[image]

realthunder, 2018-07-19:
Ah, I see. Yes, there is a known inefficiency in selection handling of object with lots of faces/edges. It have been improved. Please wait for the next release, probably early next week.

ceremcem, 2018-07-19:
Wow, great news! Thanks! (I think I can find a low detailed model for now, it doesn't have to be that drawing.)

realthunder, 2018-07-19:
Forget to say. Meanwhile, to work around this problem, just avoid select the object in tree view directly. Select in 3D view, instead, which only selects a sub-element. Most of operation in asm3 actually requires you to select element instead of the whole object.

ceremcem, 2018-08-07:
It seems that this is resolved by https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.8.0 Thanks!

closed 2018-08-07 by ceremcem