realthunder/FreeCAD_assembly3 #19: WIP: testing sympy

looooo, 2018-07-29:
some first tests with sympy (trying to constraint 2 simple boxes, point to point-constraint):
```
<asm3.gui> 122.859145 - gui.py(226): command "asm3CmdSolve" exception
Traceback (most recent call last):
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/freecad/assembly3/FCADLogger.py", line 106, in report
    return func(*args,**kargs)
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/freecad/assembly3/solver.py", line 348, in solve
    return _solve(*args,**kargs)
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/freecad/assembly3/solver.py", line 318, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/freecad/assembly3/solver.py", line 91, in __init__
    self.system.solve(group=self.group,reportFailed=reportFailed)
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/freecad/assembly3/sys_sympy.py", line 1278, in solve
    tol=algo.Tolerance,method=algo.getName(),options=algo.Options)
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/scipy/optimize/_minimize.py", line 593, in minimize
    return _minimize_powell(fun, x0, args, callback, **options)
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/scipy/optimize/optimize.py", line 2560, in _minimize_powell
    fval = squeeze(func(x))
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/scipy/optimize/optimize.py", line 293, in function_wrapper
    return function(*(wrapper_args + args))
  File "/home/k/conda/envs/fc_assembly/lib/python3.6/site-packages/freecad/assembly3/sys_sympy.py", line 1080, in F
    res = eq(*params)
TypeError: _lambdifygenerated() missing 6 required positional arguments: '_Dummy_49', '_Dummy_50', '_Dummy_51', '_Dummy_52', '_Dummy_53', and '_Dummy_54'
```

looooo, 2018-07-29:
great looking forward to test it.

luzpaz, 2021-08-05:
@realthunder this issue is being de-prioritized? (just asking)