realthunder/FreeCAD_assembly3 #20: Orientation changes independently

ceremcem, 2018-07-30:
Orientation of PCB changes out of my control: 

[image]

Files: https://github.com/ceremcem/fc-asm3-test/tree/ce2187653be3f1c1ae28d8b60c0ffeaf7d68725c

realthunder, 2018-07-30:
Can you elaborate your problem a bit more? Would do you want to achieve? I can't figure out your intention based on your screencast.

realthunder, 2018-07-30:
I guess you probably want move the cube along the plane of the PCB? If so, you need the PlaneAlignment constraint instead of PlaneCoincident. You may want to checkout the constraint introduction by forum member jpg87 [here](https://forum.freecadweb.org/viewtopic.php?f=36&t=29643)

ceremcem, 2018-07-30:
My actual intention is that the tiny cube represents an ethernet socket and underlying plane represents the PCB and it will finally be a RaspberryPI-3 model. 

I draw a pcb and an ethernet socket, then place the socket on pcb, then added a PlaneCoincident, which caused ETH socket to be placed in the center of PCB *but with a random orientation*
[image]

Then I tried to add a second plane coincident which asm3 refused. Then I tried to freely move the socket. That's the history.

realthunder, 2018-07-30:
I see. Okay, a few things. First, for any assembly, it is recommended to first create a Lock constraint to fix a base part. If there is no explicitly fixed part, the part owner of the first element in the first constraint will be fixed. In your case, this is the cube. When you try to move a fixed component, all other parts in the assembly moves together with it. If you select the plane on th PCB, however, you can rotate it as you wish.

Second, you intention is better captured by a plane alignment, which allows the cube to move freely alone the pcb plane.You can then add a second plane alignment to further restrict its position. I'd suggest you read the link in my previous post to get familiar with various constraints.

Third, for use case like PCB assembly, I suggest you create the pcb in a part design body, and attach a sketch on the pcb plane, and mark every component position with a circle. Add that body to the assembly. Then add individual components. Component placing can be easily done by a single PlaneCoincident. There is a LockAngle option in plane coincident to let you customize the rotation of the components. Or, you can use two PlaneCoincident of each component to auto calculate the rotation for you.

ceremcem, 2018-07-30:
There are 2 things I understand: 

1. This is not a bug, no need to worry about anything
2. Using PlaneAlignment did what I intended to do. I should definitely understand the various constraints with their intentions. 

Thanks.

closed 2018-07-30 by ceremcem