realthunder/FreeCAD_assembly3 #21: How to reset the orientation?

ceremcem, 2018-07-30:
When we change the orientation of an object, the top/left/right/... view tool can not show that object in original orientation. How can we reset it?

[image]

realthunder, 2018-07-30:
If you you want some kind of fixed orientation of an assembly, you should decide a base part before creating the assembly. Let's say you want the PCB to be base part. After you add the PCB to the assembly, select any face of the PCB, and click the Lock constraint (The middle one with a lock in your constraints toolbar) to fix it. You can then proceed to add other components. And when you move other components, they will always snap back to PCB after solver recompute.

You should not attempt to change the placement of the PCB, or use any of the mover tool to move the PCB, otherwise, all other components will be moved together and settle with the new PCB orientation. If you accidently move the base part, you can either click undo to undone mover change, or reset the placement by simply typing in python console

```
App.ActiveDocument.PCB.Placement = App.Placement()
```

Or, click the Lock Mover button, the middle one with a lock on the mover icon on your left toolbar. This will prevent activating mover on locked parts.

BTW, you may want to try Axial Mover, the icon below the Mover. Many user feel that it is easier to use.

ceremcem, 2018-08-07:
Locking a part at the very beginning solves the orientation issue before it takes place. That's the way I'll stick from now on. 

But anyway, I would like to be able to reset the orientation anytime. Currently, above Python code doesn't help much: 

[image]

realthunder, 2018-08-07:
Is this 'Assembly001' inside another assembly?

If you want to reset part's orientation, then you should set the part object's placement

```
App.ActiveDocument.pcb.Placement = App.Placement()
```

ceremcem, 2018-08-08:
The file I'm working on is: https://github.com/ceremcem/fc-asm3-test/tree/rpi-from-scratch [this commit](https://github.com/ceremcem/fc-asm3-test/tree/ce75a573d54ba0f3ce341e53ed069f928f3b7ddb)

Here is what I did so far:

```py
pcb = App.ActiveDocument.getObjectsByLabel('pcb')[0]
rpi3 = App.ActiveDocument.getObjectsByLabel('rpi3')[0]
eth = App.ActiveDocument.getObjectsByLabel('ethernet')[0]
fp = App.ActiveDocument.getObjectsByLabel('front-panel')[0]

pcb.Placement = App.Placement()  # Not enough on its own
rpi3.Placement = App.Placement()
eth.Placement = App.Placement()
fp.Placement = App.Placement()
```

Result is: 

[image]

closed 2018-08-08 by ceremcem