realthunder/FreeCAD_assembly3 #24: Unable to open part correctly in FreeCAD Assembly3

fpostema, 2018-08-05:
Hello realthunder,

I'm playing around with FreeCAD and Assembly3 for a few days now and I'm very impressed by what it can do. Good work. I like it!!!

While testing this software, I have noticed the following:

1. Sometimes I have to remove some constraints, because the sketch is over-constrained in preview 0.7, while it is OK in v0.17.

2. Sometimes when editing an existing sketch, the reference plane of the sketch is not correct. Try editing sketch002 of the following file: 
[ReservoirTube.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2260265/ReservoirTube.zip)

3. I have a file that I created in FreeCAD 0.17 and I would like to open it in FreeCAD + Assembly3 (preview 0.7). The attached file doesn't load correctly at all in preview 0.7. The initial shape is correct, but when the shape is recalculated, things go wrong. The problem starts with sketch002 and sketch003. In these sketches I have referenced some external geometries. It looks like v0.18 uses the wrong references.
[StoelBovenblad.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2260253/StoelBovenblad.zip)

P.S. I'm using pre-release FreeCAD-asm3-20180630-e6c14b62-f5dd9d4-Win64 on Windows 10.

realthunder, 2018-08-06:
Thanks for reporting. I can confirm this is a bug. Will fix soon.

realthunder, 2018-08-06:
New version released. Please give it a try. And about the sketch over constrained problem, that is likely caused by the modification of sketch upstream. You can try the upstream 0.18 FC and see if it has the same problem.