realthunder/FreeCAD_assembly3 #26: Can not get geometry element when underlying sketch is modified

ceremcem, 2018-08-09:
1. Drawing is at [this point]
[image]

2. Remove the small window from the sketch: 

[image]

3. `Can not get geometry element ...` exception arises:

[image]

4. That element seem to be referenced from previous state: 

[image]

realthunder, 2018-08-09:
This is by design. Because you remove the window, and created another one, the face is consider a different one. To avoid this type of problem, when you use PartDesign body as part, you should choose shape in history as early as possible to avoid lost of element reference.

To fix the error here, select the new face (preferably from early body shape history), and drag and drop to the element in error.

ceremcem, 2018-08-09:
Is that what you are suggesting:

[image]

realthunder, 2018-08-09:
The face dragging is correct. But you don't need to delete that element. Just drop the face to that element can fix the problem. 

Another thing, I mentioned that to avoid similar problems from happening again, it is better to use early feature. It appears that sketch is the first feature in your body, in which case, you can use the origin plane that your sketch is attached to for constraining.

ceremcem, 2018-08-26:
I'm slowly getting the hang of the internal working of the event: When we create a constraint between two "elements" and if we point to "that square face of the box", it gets the square face of the box. If we open a pocket in the box, then the face gets another id and naturally the link can not find appropriate face. 

Here we are assembling two objects: 

[image]

When we create a pocket, the element link says it can not find `...Face6`:

[image]

Which means at the beginning, that face was enumerated as `Face6`.  To fix the matter, we click the face again, select the highlighted line in the treeview, drag and drop it on the broken element: 

[image]

Okay, we are seeking for trouble, so remove the pocket:

[image]

Now element says it can not find `...Face10`, because the windowed face was (apparently) enumerated as `Face10`. The enumeration is kind of deterministic, because when we fix this issue in the same way we did before and re-open the pocket, it can't again find `Face6`. 

But if we flip the gray part and perform the same operations, element link **won't be broken**, because that face is always enumerated *before* pocket faces, so its ID is always the same (like `Face4`). 

# Conclusion 

There is no reliable way to prevent this problem before it happens, because we might have to create two distinct constraints where one of them relies one side of the gray object and the other constraint relies on the other side of the gray object and then we might want to change the underlying sketch. 

# Proposal

If everything is true above, then I think the way to go is leaving already enumerated faces (and edges) intact and giving new IDs for new faces/edges.

closed 2018-08-26 by ceremcem

ceremcem, 2018-08-26:
Now I understand what you mean by "...it is better to use early feature.". I just used pocket feature and now I have actual history. 

Thanks!