realthunder/FreeCAD_assembly3 #27: Can not delete the duplicate object

ceremcem, 2018-08-12:
Attempt to remove the somehow duplicated object causes deletion of the original: https://github.com/ceremcem/fc-asm3-cut-window/issues/1

maidenone, 2018-08-14:
`
Traceback (most recent call last):
  File "/home/maiden/projects/FreeCADLINK/build/Ext/freecad/asm3/assembly.py", line 353, in execute
    self.getElementSubname())
  File "/home/maiden/projects/FreeCADLINK/build/Ext/freecad/asm3/assembly.py", line 897, in getElementInfo
    part.Name,subname))
<type 'exceptions.RuntimeError'>: cannot get geometry element from Body001.Pad001.;#245;:G2#247;XTR;:T1736:4:F.Face2
Traceback (most recent call last):
  File "/home/maiden/projects/FreeCADLINK/build/Ext/freecad/asm3/assembly.py", line 353, in execute
    self.getElementSubname())
  File "/home/maiden/projects/FreeCADLINK/build/Ext/freecad/asm3/assembly.py", line 897, in getElementInfo
    part.Name,subname))
<type 'exceptions.RuntimeError'>: cannot get geometry element from Body001.Pad001.;#245;:G3#249;XTR;:T1736:4:F.Face3
`
[image]

ceremcem, 2018-08-14:
Elements doesn't seem to be broken on my side:
[image]

Are we using the same commit?

ceremcem, 2018-08-15:
This issue may or may not be related with asm3 workbench. I'm closing the issue because [I'm not able to reproduce](https://github.com/ceremcem/fc-asm3-cut-window/issues/1#issuecomment-412377408) it.

closed 2018-08-15 by ceremcem