realthunder/FreeCAD_assembly3 #29: Is it possible to overwrite some properties of linked object?

ceremcem, 2018-08-15:
As for as I could understand, linked objects' properties are exactly same of their parent (class) object, except for `Placement` property (they can be placed independently). 

Likewise, can we overwrite some other properties of a linked object, like `Placement`? 

### Rationale 

I have a bent sheet metal and I want to generate 2D drawing of unfolded form of it to be able to generate laser cutting machine drawings. For that purpose, I created a linked copy and unfolded it, generated the drawing; but original bent form is unfolded too. I can create a copy and unfold it, but I want my further changes to be reflected to the unfolded instance (such as placement of socket window cuts). 

So, can I overwrite some properties of a linked object (`'Parameters Ex'.'unfold'` in this case)?

realthunder, 2018-08-15:
Link does not know how the linked object generate its shape, so it can't be used for this purpose. But there might be other way to achieve what you want. Can you please attach some sample file here. I'll see what I can do.

ceremcem, 2018-08-16:
Sorry for being late, lots of things happen on other sides. 

Here is a screen cast that demonstrates what I want to achieve (partly, of course): 

[image]

What I want to achieve is: 

* After creating link, I want to create a drawing according to the *linked object*, not the original (bent) one. 
* When I set `unfold` to `true`, I only want to see the linked object as unfolded, not the original one. 

[Here is the model](https://github.com/ceremcem/fc-asm3-test/blob/df08e42f3ecee7da49b571e35c889d7ff47e5a15/assembly.fcstd)

ceremcem, 2018-08-18:
@realthunder Currently this particular use case [is resolved](https://github.com/shaise/FreeCAD_SheetMetal/issues/38) and I can't imagine this (overriding some properties) is useful in which circumstances, so I'm going to close the issue for now.

closed 2018-08-18 by ceremcem