realthunder/FreeCAD_assembly3 #32: Can not change/delete sketch constraint

ceremcem, 2018-08-19:
### Problem 

Can not change height of the enclosure (currently 77.44mm)

[image]

### Reproduction 

1. Drawing is [here](https://github.com/ceremcem/fc-asm3-test/blob/ec0d43f25cd6065a38d7f8897faa167039d688a6/assembly.fcstd)

2. Navigate to the panel/pad/sketch

3. Try to delete the `|--| 77.44mm` constraint

4. Exception arises as `Property not found`

realthunder, 2018-08-20:
This an upstream bug. I have submitted a [PR](https://github.com/FreeCAD/FreeCAD/pull/1619). I'll include the fix in my next release.

ceremcem, 2018-08-22:
Confirm fixed since https://github.com/realthunder/FreeCAD_assembly3/releases/tag/0.8.5

closed 2018-08-22 by ceremcem