realthunder/FreeCAD_assembly3 #33: addParamV() exception (Build Question?)

ceremcem, 2018-08-22:
I've built LinkStage3 by myself and here is its "About" info:

```
OS: Debian GNU/Linux 8.4 (jessie)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14929 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: 87a1f78a31b5c1ff5e64838c82e79dc8963107c9
Python version: 2.7.9
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

Constraint solver is [SolveSpace](https://github.com/realthunder/FreeCAD_assembly3/wiki/Build-Instruction#solvespace) (which is also built by me)

Here is simple assembly I've tried: 

[image]

Error traceback is: 

```console
<asm3.gui> 18.47175 - gui.py(261): command "asm3CmdSolve" exception
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/FCADLogger.py", line 106, in report
    return func(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 404, in solve
    return _solve(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 374, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 49, in __init__
    self.v0 = self.system.addParamV(0,group=self._fixedGroup)
TypeError: addParamV() got an unexpected keyword argument 'group'
```

Note that your 0.8.4 release has no such issue. Is it wrong approach to build against latest LinkStage3 commit?

realthunder, 2018-08-22:
That's an unexpected error. What is the swig version you are using? Please attach your py_slvs/slvs.py file here.

ceremcem, 2018-08-22:
```console
# swig -version
SWIG Version 2.0.12
Compiled with g++ [x86_64-unknown-linux-gnu]
Configured options: +pcre
```

[py_slvs/slvs.py](https://github.com/ceremcem/fc-asm3-test/blob/issue-33/slvs.py)

realthunder, 2018-08-22:
Please use swig 3. I am using 3.0.8.

ceremcem, 2018-08-22:
I'm utterly confused with the code names of Debian releases, I'm upgrading from old-stable (jessie) to stable (stretch) which in turn will upgrade swig to 3.

ceremcem, 2018-08-22:
Working perfectly. Thanks!

closed 2018-08-22 by ceremcem