realthunder/FreeCAD_assembly3 #35: Cyclic Dependency exception while creating sub-shape binder

ceremcem, 2018-08-30:
When I'm trying to follow the [Part Workbench section of Modeling using Assembly](https://github.com/realthunder/FreeCAD_assembly3/wiki/Modeling-using-Assembly#using-part-workbench) tutorial, I'm getting "Cyclic dependency" exception while creating the sub-shape binder: 

[image]

Here is the model I've created: [test.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2334743/test.zip)

ceremcem, 2018-08-30:
I *think* this is because the sub-shape binder wants to go under the first `Body` part, in this case, this is `socketpart.Body` and if `Assembly.Parts` keeps everything as a feature list/history list (I might
 name this wrongly), then as the sub-shape binder is tried to be created under a previous part in the history, it may perfectly throw a cyclic dependency exception: 

[image]

realthunder, 2018-08-30:
This is a bug. I'll fix it soon.

realthunder, 2018-08-31:
New version released, please check.

ceremcem, 2018-08-31:
Works like a charm! Thanks!

closed 2018-08-31 by ceremcem