realthunder/FreeCAD_assembly3 #36: Problem while moving calculated parts

ceremcem, 2018-08-31:
# Description 

I defined a cutter shape, defined a panel, created a sub-shape binder and cut the panel with the binder. That works and updates work only if I move the original panel, not the cut version: 

[image]

Here is the model: [test2.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2339283/test2.zip)

# Expected Behaviour 

Since I need to move what I'm seeing, I should be able to move the `Cut`, rather than `panel`.

realthunder, 2018-08-31:
I explained this use case in the tutorial. You are using the Part Workbench Cut approach, hence you should only apply assembly constraint (same for part move) on the uncut panel. In fact, this workflow will not work if you have any movement in the cut feature. In other word, the cut feature should always have a null Placement.

ceremcem, 2018-08-31:
Then what might the purpose (use case) of moving the `Cut` be at this point of FreeCAD(-asm3)? If there is no use case for current behaviour, I *think* moving the `Cut` should just translate the placement of its `Base` transparently, or shouldn't let movement at all. What do you think?

realthunder, 2018-08-31:
Well, this so called _design in context_ use case is very new to FC. The Part Cut feature is certainly not aware of this possible work flow. Right now, users must discipline themselves if they are willing to venture into this new territory.

> I think moving the Cut should just translate the placement of its Base transparently, or **shouldn't let movement at all. What do you think?**

asm3 offers some minimum movement protection, but requires explicit user setup. You can lock the Cut feature, and then turn on the [Lock Mover](../blob/master/Gui/Resources/icons/Assembly_LockMover.svg) option. This will prevent you from accidentally move the part using mover.

ceremcem, 2018-08-31:
Okay, at the moment I'll use the workflow with all the workarounds and it's up to you to keep this issue open or closed because I'm willing to implement such a feature. I guess it would also be a good starting point. 

Lock Mover will definitely help in my work, but the main purpose for the proposal was that it was confusing at the beginning, so as `Cut` would never be moved, I thought the lock feature (or any exception message) might be implemented into the Asm3. 

Now I know how to deal with this, so this is not urgent to me.