realthunder/FreeCAD_assembly3 #40: FreeCAD crash

ceremcem, 2018-08-31:
I don't know if this is somehow related with LinkStage3 branch or not, here is the most recent traceback output before the crash: 

```
<Gui::Document> Document.cpp(367): cannot edit without active view
unfold successful
unfold successful
Exception (Fri Aug 31 22:37:58 2018): no view  
Exception (Fri Aug 31 22:38:10 2018): no view  
Exception (Fri Aug 31 22:38:13 2018): no view  
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x35fc0) [0x7f8d8a26afc0]
#1  0x7f8d137b2d62 in PartGui::TaskAttacher::onSelectionChanged(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/PartGui.so+0x162
#2  0x7f8d8cc9108f in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0xf
#3  0x7f8d8cca5a57 in boost::signal1<void, Gui::SelectionChanges const&, boost::last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)> >::operator()(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x507
#4  0x7f8d8cc94d27 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x1f7
#5  0x7f8d8cca5a57 in boost::signal1<void, Gui::SelectionChanges const&, boost::last_value<void>, int, std::less<int>, boost::function<void (Gui::SelectionChanges const&)> >::operator()(Gui::SelectionChanges const&) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x507
#6  0x7f8d8cca0798 in Gui::SelectionSingleton::addSelection(char const*, char const*, char const*, float, float, float, std::vector<Gui::SelectionSingleton::SelObj, std::allocator<Gui::SelectionSingleton::SelObj> > const*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x808
#7  0x7f8d8cafa3a3 in Gui::SoFCUnifiedSelection::setSelection(std::vector<Gui::SoFCUnifiedSelection::PickedInfo, std::allocator<Gui::SoFCUnifiedSelection::PickedInfo> > const&, bool) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x6a3
#8  0x7f8d8cafb833 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x113
#9  0x7f8d89bbdc0c in SoNode::handleEventS(SoAction*, SoNode*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x4c
#10  0x7f8d89983c04 in SoAction::traverse(SoNode*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x54
#11  0x7f8d89ad3294 in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x174
#12  0x7f8d89bb366f in SoGroup::doAction(SoAction*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x3f
#13  0x7f8d89bd06d7 in SoSeparator::doAction(SoAction*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x27
#14  0x7f8d89bbdc0c in SoNode::handleEventS(SoAction*, SoNode*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x4c
#15  0x7f8d89983c04 in SoAction::traverse(SoNode*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x54
#16  0x7f8d8998e1aa in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x7a
#17  0x7f8d899847d2 in SoAction::apply(SoNode*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0xd2
#18  0x7f8d89af10a9 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libCoin.so.80+0x69
#19  0x7f8d8cb6a0de in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x1e
#20  0x7f8d8cb68283 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x43
#21  0x7f8d8cb9b7e9 in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x9
#22  0x7f8d8cb80b45 in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x215
#23  0x7f8d8cb7c162 in Gui::NavigationStyle::processEvent(SoEvent const*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x62
#24  0x7f8d8cb9b798 in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x88
#25  0x7f8d8cb698dc in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x16c
#26  0x7f8d8a75b646 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x86
#27  0x7f8d8ac8ce0c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x6c
#28  0x7f8d8ac935dd in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x3ad
#29  0x7f8d8c9856d1 in Gui::GUIApplication::notify(QObject*, QEvent*) from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x41
#30  0x7f8d8a75b4dd in QCoreApplication::notifyInternal(QObject*, QEvent*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x6d
#31  0x7f8d8ac92d93 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x153
#32  /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2449cb) [0x7f8d8ad079cb]
#33  0x7f8d8ad07269 in QApplication::x11ProcessEvent(_XEvent*) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1559
#34  /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26bb02) [0x7f8d8ad2eb02]
#35  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2e7) [0x7f8d84276287]
#36  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4c4c0) [0x7f8d842764c0]
#37  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f8d8427654c]
#38  0x7f8d8a7887a1 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x71
#39  /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26bbb6) [0x7f8d8ad2ebb6]
#40  0x7f8d8a75a0af in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x2f
#41  0x7f8d8a75a3a5 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x175
#42  0x7f8d8a75fb79 in QCoreApplication::exec() from /tmp/.mount_freeca9Qr3fD/usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x89
#43  0x7f8d8c93e303 in Gui::Application::runApplication() from ././/lib/freecad-daily/lib/libFreeCADGui.so+0x16a3
#44  freecad-daily(main+0x74b) [0x40355b]
#45  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f8d8a257b17]
#46  freecad-daily() [0x4044c7]
```

The last thing I was doing is as follows: 

I changed some dimensions via Spreadsheet. Then went back in the history, then the crash happened.

realthunder, 2018-09-01:
I did find something potentially wrong in TaskAttacher onSelectionChanged(), not sure if it is related.

realthunder, 2018-09-02:
Should be fixed in LinkStage3

ceremcem, 2018-09-02:
Since I can not provide a reproduction procedure, I'm going to close this issue without being able to confirm.

closed 2018-09-02 by ceremcem