realthunder/FreeCAD_assembly3 #42: Same label can be assigned in treeview

ceremcem, 2018-09-01:
In upstream, same label is not allowed in any case: 

[image]

In LinkStage3, we can assign the same label in treeview erroneously: 

[image]

# Reproduction 

1. Add a part, name it "myobject"
2. Add another part, try to name it "myobject"
3. The second part will be named as "myobject001"
4. Try to rename "myobject001" as "myobject" again
5. See the second object is renamed "succesfully" as "myobject" *only* in the treeview

realthunder, 2018-09-01:
Confirmed. Will fix soon.

realthunder, 2018-09-02:
Fixed in LinkStage3.

closed 2018-09-02 by ceremcem