realthunder/FreeCAD_assembly3 #43: Measurement WB throws exception

ceremcem, 2018-09-01:
Measurement WB works correctly on my simple tests, I can not reproduce the exception on newly created drawings but here is the traceback: 

```console 
Traceback (most recent call last):
  File "/home/ceremcem/.FreeCAD/Mod/Manipulator/Caliper.py", line 382, in addSelection
    plcm, top_level_obj, bbC, pnt, orient, norm = get_placement_hierarchy (sel[0])
  File "/home/ceremcem/.FreeCAD/Mod/Manipulator/Caliper.py", line 1117, in get_placement_hierarchy
    listSorted=get_sorted_list (Obj)
  File "/home/ceremcem/.FreeCAD/Mod/Manipulator/Caliper.py", line 1026, in get_sorted_list
    if len(ap.InListRecursive) < lvl:
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'InListRecursive'
Traceback (most recent call last):
  File "/home/ceremcem/.FreeCAD/Mod/Manipulator/Caliper.py", line 382, in addSelection
    plcm, top_level_obj, bbC, pnt, orient, norm = get_placement_hierarchy (sel[0])
  File "/home/ceremcem/.FreeCAD/Mod/Manipulator/Caliper.py", line 1117, in get_placement_hierarchy
    listSorted=get_sorted_list (Obj)
  File "/home/ceremcem/.FreeCAD/Mod/Manipulator/Caliper.py", line 1026, in get_sorted_list
    if len(ap.InListRecursive) < lvl:
<type 'exceptions.AttributeError'>: 'NoneType' object has no attribute 'InListRecursive'
```

for the following operation: 

[image]

In this model: [rpi3.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2342576/rpi3.fcstd.remove-extension.zip)

easyw, 2018-09-01:
Measurement WB doesn't even exist. 😝

ceremcem, 2018-09-01:
What?! :D How come?
[image]

easyw, 2018-09-01:
> What?! :D How come?

Who knows ... If you can find Measurement WB you are lucky... but I can only find Manipulator WB 😉

ceremcem, 2018-09-01:
AAAAAAAAAAAAAAHhh :D

easyw, 2018-09-01:
1) your are using MWB on A3 branch... this is not fully supported ATM... I'm slowly including the new containers that A3 is using... 
At the same time, which is your MWB release and particularly Caliper release? 
I cannot reproduce your issue witht the attached file on Caliper v1.43 and A3 0.86

ceremcem, 2018-09-01:
When I switch to MWB, Report view says its 
```
Manipulator WB Activated(v 1.2.6)
```
When I click the "Caliper", Report view says 
```
Caliper version 1.4.3
```

I'm also using A3 0.8.6 from realthunder's releases on Debian, with Appimage.

ceremcem, 2018-09-01:
Wow, yes, I also can not reproduce my issue with my own file when I download it separately. Weird. Let me find the cause.

ceremcem, 2018-09-01:
Okay, here is the reproduction steps: 

1. Download [test3.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2342670/test3.zip)
2. Open `enclosure.fcstd` (not `rpi.fcstd` directly)
3. This will automatically open `rpi.fcstd`
4. Try to measure the distance between holes

easyw, 2018-09-01:
ok, I can reproduce it
BTW the doc is complex... it is using external docs/links... 
I will have a look at...

ceremcem, 2021-02-21:
Closing as it's already belongs to an external workbench and I can not reproduce the problem.

closed 2021-02-21 by ceremcem