realthunder/FreeCAD_assembly3 #44: Parts are moving together unexpectedly

ceremcem, 2018-09-01:
Here the `rpi` and `Extrude` parts are in the same hierarchy, and `Auto recompute` is disabled, so I would expect to be able to move the `Extrude` independently, but it is bound to the `rpi`: 

[image]

Model is [here](https://github.com/realthunder/FreeCAD_assembly3/files/2342670/test3.zip)

realthunder, 2018-09-01:
Because there are multiple levels hierarchies in-between, the part mover defaults to move parts in the first hierarchy, therefore, you are actually moving rpi-with-env. To move Extrude inside rpi-with-env, you need to add rpi-with-env as the second selection before activating mover.

ceremcem, 2018-09-01:
Can you be more specific? What do you mean by "adding an object/container as second selection"?

ceremcem, 2018-09-02:
I can't believe myself. Sorry for the duplicate and thank you again. (I added this into [my notes](https://github.com/ceremcem/freecad-notes/blob/master/README.md#by-task-asm3))

closed 2018-09-02 by ceremcem