realthunder/FreeCAD_assembly3 #45: Can't create a sketch every time on the clicked surface

ceremcem, 2018-09-01:
I'm expecting to be able to create a sketch right on the surface when I click the surface at any time:

[image]

But when I move the linked part, I can not draw on the surface anymore: 

[image]

Model: [sketcher-test-external.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2342701/sketcher-test-external.zip)

realthunder, 2018-09-01:
Because the object is external, sketch does not support external mapping. Right now, my branch only support external object in-place-editing, not creating. You can use the navigation [button](https://github.com/realthunder/FreeCAD/blob/LinkStage3/src/Gui/Icons/LinkSelect.svg) to jump to the document containing the object, and create a sketch there. Once finished, drag the  object into the assembly.

BTW, if you stick to PartDesign, then it is possible to create new sketch in-place. Because PartDesign has a concept of active body. You can double click to select an active body, and new sketch will be added to that body, and it works even if the body is external.

ceremcem, 2018-09-01:
Creating sketch in the original document may or may not help, and is not helping in this case: 

[image]

If the `pcb` Placement is reset, sketch is created in the correct place. 

> Second and off-topic, I was not aware of "in place editing support". That sounds something I should avoid using it by mistake. I was thinking the external objects are handled just like any internal object where only the geometry is gathered from the external file and its placement is managed by the current project.

realthunder, 2018-09-02:
> If the pcb Placement is reset, sketch is created in the correct place.

If you move the sketch into the assembly, then it will be in the correct position. But I can see this is confusing to the user. I'll add some support of in-place-creation for sketch.

> Second and off-topic, I was not aware of "in place editing support". That sounds something I should avoid using it by mistake.

The same object can now appear in multiple places through linking. In-place-editing means to edit the object at the placement where you launch the editing (e.g. by double clicking).

ceremcem, 2018-09-02:
> You can edit the object through a link. You can try link to a sketch and edit it by double click the link.

Wow, yes! Thank you for explicitly indicating this:

[image]

> And why not use git on your files

I spent many hours on this. There is a success to some extent but `git submodules` can not be used effectively for the following reasons: 

1. Since everything is stored in binary format, if you have a big external object (Raspberry model, for example), as FreeCAD updates some metadata in the binary every time, every commit will bloat the history, which will be unmanageable in time. I even tried [unpacking and re-packing](https://forum.freecadweb.org/viewtopic.php?f=10&t=30267&sid=4ee7870825806aa8e940b349d848a291&start=10#p249896) which does't seem to help reducing the diff size.

2. Sharing a model with sub-modules will involve conflict resolution. Since it is impossible to perform such thing  by looking at the diff of `fcstd` files right now, it requires an extensive amount of attention and consumes a great amount of brain power while editing the shared model, since you have to remember your changes and explicitly describe in your commit, because there is no other possible way to know what is changed in two versions of a shared object and versions *will change* in every commit. [Here is an idea](https://forum.freecadweb.org/viewtopic.php?f=8&t=30666) to solve this one. 

I'm currently using git on my projects and discarding the git history when I'm done.

realthunder, 2018-09-02:
> Since everything is stored in binary format

On the contrary, FC stores almost everything in text format. The fcstd file is nothing but a zip archive, and that's why I mentioned to disable compression when adding to git, using Menu->Preferences->Document->Compression level, set to 0.

> Since it is impossible to perform such thing by looking at the diff of fcstd files right now, it requires an extensive amount of attention and consumes a great amount of brain power while editing the shared model, since you have to remember your changes and explicitly describe in your commit

This is largely true. I think this applies to anything that is not handwritten. And just like writing a code library that is shared by others, you define interfaces. And asm3 lets you do similar thing, using [Elements](https://github.com/realthunder/FreeCAD_assembly3/wiki/Concepts#element), which can be considered as an abstraction layer for the geometry elements. For other models, you can parameterize your design as much as possible using spreadsheets, and put detail documents there. I think this applies to all non handwritten designs, be it CAD, PCB or whatever, that you compensate the lack of meaningful diff with detail documents.

ceremcem, 2018-09-06:
> The fcstd file is nothing but a zip archive 

Which is a binary format in the end. Git will add the whole zip file in every commit, thus bloat the history in size. Here is a related [SO question](https://stackoverflow.com/questions/8001663/can-git-treat-zip-files-as-directories-and-files-inside-the-zip-as-blobs). I'm not sure how setting the compression level to 0 would help, btw. 

Returning to main objective, creating a sketch on a selected surface seems impossible right now, either in the end product or in the owner file. For example, I'm trying to create a sketch on the surface, but it is actually created in the middle of the device:

[image]

realthunder, 2018-09-06:
> Which is a binary format in the end.

Yeah, you're right. I just checked, the git diff of uncompressed zip shows nothing. But I would imagine it's not that difficult to add an option to save the document as folder instead of a single file.

> For example, I'm trying to create a sketch on the surface, but it is actually created in the middle of the device

I'll improve sketch creation command in the future. Meanwhile, after you have created the sketch, manually move the sketch to the same container as its support object, **with the new sync placement option DISABLED**. The sketch will be at the correct position.

ceremcem, 2018-09-06:
> But I would imagine it's not that difficult to add an option to save the document as folder instead of a single file.

That would be a nice enhancement.

Isn't your suggestion the following: 

[image]

realthunder, 2018-09-06:
What is your usage of the sketch? To do further modeling in the body? If so, you should go to PartDesign WB, and create sketch there.

ceremcem, 2018-09-06:
That will be the `socket-cuts` in the end. I'm trying to place the sketch right in front of the sockets because when I'll edit, I need to see the edges (they are overlapped by model at the moment)

realthunder, 2018-09-06:
Oh, and you may want to reset ethernet body's placement before creating the sketch and stuff.

ceremcem, 2018-09-06:
> Then you need to change the grouping

Think this RPi model is an external model and I shouldn't edit it (unless I want to create a different version based on that one). 

My current workaround is shifting the sketch base: 

[image]

realthunder, 2018-09-06:
In that case, you need to use the SubShapeBinder to bring the ethernet object to the global coordinate space. Before creating the SubShapeBinder, make sure you don't have any active PartDesign body, otherwise, the binder will be added that body. If it happens, simply drag the binder out, and update its reference. Then you can create sketch on the binder.

ceremcem, 2018-09-28:
Using sub-shape binder perfectly solves the issue. Overall procedure: 

1. Create a sub-shape binder in your desired active body 
2. Hide the original object 
3. Click on desired surface on the binder
4. Click the "Create Sketch" button

ceremcem, 2018-09-28:
> But I would imagine it's not that difficult to add an option to save the document as folder instead of a single file

I want to touch this topic again. Is that really not too difficult to implement? Besides many advantages, it would help investigating bugs, like #76

closed 2018-09-28 by ceremcem

ceremcem, 2018-09-28:
> How can it help?

Thinking again, this can be achieved by simply unpacking the project file into a temporary directory on every `git commit`, but I'd like to share what I'm thinking: I guess these lost external geometries have to be stored in the project file. There should be some lines missing or some addition/deletion might invalidate these external geometries. I thought I could have an X-Ray vision with `git diff` so I can spot any lost constraints or something right it is lost, not when I double click the sketch.

realthunder, 2018-09-28:
I see what you mean. I'll take on the sketch problem next. It is easy to handle this (I mean marking an error on lost of geometry) inside the program rather than burden user with it. But it takes some time to add the feature to allow sketch to preserve the lost geometry.

ceremcem, 2018-09-28:
> I mean marking an error on lost of geometry 

That would help very much since it will help me determining exactly when the "lost geometry incident" is happened, thus the reason of the problem. 

> But it takes some time to add the feature to allow sketch to preserve the lost geometry.

I think this is what you described in #90