realthunder/FreeCAD_assembly3 #47: Using labels on Github issues

ceremcem, 2018-09-02:
@realthunder would you mind using labels for the issues because that also helps querying later for us.

realthunder, 2018-09-02:
Sure.

closed 2018-09-02 by realthunder

ceremcem, 2018-09-07:
I suggest `upstream` label for [this](https://github.com/realthunder/FreeCAD_assembly3/issues/69) kind of issues.

ceremcem, 2018-09-07:
I suggest `upstream/merge` label for [these](https://github.com/realthunder/FreeCAD_assembly3/issues/69) kind of issues.

* `upstream/merge` means there is patches already in the upstream, waiting to be merged into LinkStage3
* `upstream/issue` this issue is related with upstream and should be solved in the upstream 
* `upstream/forum` this issue has nothing to do with LinkStage3 and probably a newbie question, should be found in the forum