realthunder/FreeCAD_assembly3 #48: "Select linked object" jumps to deepest object, not to the owner

ceremcem, 2018-09-02:
"Select linked object" button jumps to the deepest object, not the immediate owner/parent:

[image]

Instead, I would expect it to select the nut in the parent external object, `rpi-with-env` in this case: 

[image]

While I'm not able to reproduce, the "Select the deepest linked object" seems to jump to the owner, not the deepest.

realthunder, 2018-09-02:
Pad is not a link per se. The little arrow on the right bottom of the icon means it is an external object brought in indirectly by some parent link. I will change the link select command to enumerate on parent link as well.

closed 2018-09-05 by realthunder

ceremcem, 2018-09-05:
Nicely working!