realthunder/FreeCAD_assembly3 #49: linked circular edge inside a sketch not used in pad

wsteffe, 2018-09-02:
I have reproduced the problem in the model uploaded into "Freecad Forum/Assembly/Assembly3 Preview". The model contains 2 sketches and 2 pads.

Each sketch inludes a rectangle and an inner circle.

In the first sketch the circle is created with circle feature of sketch.
In the second sketch the circle is created as a link to external geometry (the hole created by padding the first sketch).

Both sketches are paded but the result is different:
-The first sketch produces a brick with a hole (as expected)
-The second sketch produces only a brick without the hole.

[image]
[image]
[image]

ceremcem, 2018-09-02:
[(Continuing from FreeCAD forum)](https://forum.freecadweb.org/viewtopic.php?f=20&t=25712&p=254021#p253966)

> quote from wsteffe:
> Actually I do not understand your image because the  cylinder with diameter=10 mm is a pad (not a 
 hole) like the rectangular box.
> Why in your image there is a discontinuity between that cylinder and the box ? They should be fused.
> 

It's maybe because I created that image with Gimp, by copy/paste :P 

As it is already answered in [the forum](https://forum.freecadweb.org/viewtopic.php?f=20&t=25712&start=890#p254028), there is no circle in `Sketch001` thus `Pad001`. 

If you want to explode it without using Asm3 features (if you don't want something dynamic) you should perform a pocket operation. This is the result: 

[image]

This is the file: [aaa.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2343726/aaa.fcstd.remove-extension.zip)

If you want something dynamic, you should see [Modeling Using Assembly](https://github.com/realthunder/FreeCAD_assembly3/wiki/Modeling-using-Assembly) tutorial.

realthunder, 2018-09-02:
> In the first sketch the circle is created with circle feature of sketch.
> In the second sketch the circle is created as a link to external geometry (the hole created by padding the first sketch).

In current FC, external geometry can only be used to constrain other geometry you created. In your specific case, you have to manually create a circle constrained to external circle's origin and radius. 

There is pull request in upstream to let use the external geometry for modeling, it is not merged yet.

wsteffe, 2018-09-03:
Thanks realthunder, 
 I think you are referring to the pull request named "Sketcher: External Geometry as defining" of abdullahtahiriyo. I have lloked into the realted forum thread but it seems that this work was halted due to problems related with the proposed extension of ExternalProperty Class.
I hope that a viable solution will be found soon.

luzpaz, 2021-03-03:
> There is pull request in upstream to let use the external geometry for modeling, it is not merged yet.

@realthunder which one?

realthunder, 2021-03-03:
I think there was a PR for that almost three years ago, before I worked on topo naming.