realthunder/FreeCAD_assembly3 #50: Toggling visibility may become broken in certain cases

ceremcem, 2018-09-02:
See "Broken Visibility Toggling" title in https://github.com/realthunder/FreeCAD_assembly3/issues/39#issuecomment-417944345

Counterfeiter, 2022-01-15:
Hello,

I have the same problem. But I am not in an assembly container and your workaround fails. If I open it in the master freeCAD branch. The exported sub sketches function will fail (sure), but the visibility is fixed.

If I understand it correctly it is the case with different coordinate systems. Is there a second workaround so get my sketch back working? I am to new to know a good solution.

So, I came a long way here. This freeCAD branch works better then the master branch, but I failed again (sixed try) to get my sketch done without break it completely. 

OS: Ubuntu 20.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

[DobsonEQPlatformClean.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7874752/DobsonEQPlatformClean.zip)

Platformtop and Platformbottom could not switched to invisible. Also the wire mode is without function for this two objects.

luzpaz, 2022-04-18:
@ceremcem please retest when you have a moment? Thanks!

luzpaz, 2022-06-30:
@ceremcem ping