realthunder/FreeCAD_assembly3 #51: Center to plane constraint example

ceremcem, 2018-09-03:
Is it currently possible to create a constraint between the center of the object's edge with a plane? This need is raised while creating a drawing from the following blueprint:

[image]

The closest option seems `MidPoint` constraint, but I couldn't make it work this way.

realthunder, 2018-09-03:
I just push a fix for MidPoint constraint. However, if don't need this constraint here. Is this what you want?

[image]

ceremcem, 2018-09-03:
No, I think I totally misunderstood the `MidPoint` constraint (I didn't use it either). 

It's something like the following, only think that at the first click we could select the red point: 

[image]

[image]

realthunder, 2018-09-03:
> No, I think I totally misunderstood the MidPoint constraint (I couldn't use it either).

Have you synced my asm3 repo?

[image]

realthunder, 2018-09-03:
Set the `Offset` property of the plane ElementLink object.

ceremcem, 2018-09-03:
Thanks! 

Here is the overall example: 

[image]

closed 2018-09-03 by ceremcem

ceremcem, 2018-09-04:
CenterToPlane operation is used so often that it becomes a burden to place a DraftPoint, name it, move to assembly every time. Would you consider adding a constraint for that purpose?

realthunder, 2018-09-04:
You don't need the draft point or the MidLine constraint at all. You can directly use the edge as the 'Point' in the PointPlaneDistance constraint. Please checkout [here](../wiki/Constraints-and-Solvers#constraining-geometry-element) about various interpretations of a geometry element.

ceremcem, 2018-09-05:
In the page you referred, an edge is considered as a point by its middle point and that would solve this particular case nicely. But I think there might be a problem, because PointPlaneDistance constraint is disabled when I click the edge:
[image]

ceremcem, 2018-09-06:
Nicely works: 

[image]

commit dda56db7 (2018-09-05): utils: relax hasCenter()

Fixes #51