realthunder/FreeCAD_assembly3 #53: Centering a point and a line with minimum distance

ceremcem, 2018-09-04:
I needed to place the SD Card in the middle of Raspberry. The flow I could think of is: 

* Place a draft point in the middle of SD Card to mark the middle point. 
* Create a draft plane in the middle of PCB
* Use PointOnLine coincident. 

[image]

How can I create such a DraftPlane?

realthunder, 2018-09-04:
Same goes for #51, you can simply use PlaneCoincident to center coincide the PCB and the SD card. 

Remember I suggested several times about the master sketch approach? In reality, when you mount something to PCB, you must have mounting holes, or soldering pads. You should use a master sketch to define those things first, which is easy if you already have the blueprint. If not, then the sketch serve the purpose of producing that blue print. After the sketch is done, create a sub-assembly to wrap the PCB with the sketch, and insert the sub-assembly to the main assembly. You can repeat this process for every  electronic components, especially for those with irregular shapes, e.g. to use a sketch circle to mark the SD card mounting pin. Then the final assembling is simply to constrain each corresponding pair of sketch circles with PlaneCoincidence.

ceremcem, 2018-09-06:
>  You should use a master sketch to define those things first, which is easy if you already have the blueprint.

I'm generally creating a PCB from scratch and the first step is fitting it into the mechanical environment. In the end, I want to achieve the following workflow while designing the devices: 

- [x] Create PCB with appropriate edge cuts according to the mechanical environment [[1]](https://github.com/realthunder/FreeCAD_assembly3/issues/63) 
- [x] Place the components freely on the PCB which will satisfy the environmental requirements, user experience, aesthetic etc.
- [x] When 3D design is finished, let the electronic components "drill" their own holes dynamically [[2]](https://github.com/realthunder/FreeCAD_assembly3/issues/25)
- [ ] Let FreeCAD generate PCB edge cuts and soldering pads automatically from 3D design (like, if I guess, what StepUp does) as DXF and import it from the PCB software
- [ ] And as a side project, I'm planning to create a real time communication between FreeCAD and the PCB software (KiCAD, AeCAD, etc...) to synchronize those changes in real time. [[3]](https://github.com/aktos-io/aktos-dcs)

You are suggesting the opposite, or am I still misunderstanding?

>  you can simply use PlaneCoincident to center coincide the PCB and the SD card

The actual goal is to center the SD Card according to the PCB, but they are not on the same planes:

**File**: [centering-planes.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2358811/centering-planes.fcstd.remove-extension.zip)

[image]

Currently I can achieve this goal with the combination of the following constraints: 

[image]

**File:** [centering-planes-2.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2358827/centering-planes-2.fcstd.remove-extension.zip)

If there is not a misunderstanding about the constraints on my side, I may clarify my proposal.

realthunder, 2018-09-07:
> Create PCB with appropriate edge cuts according to the mechanical environment [1]

This is step is fine, after which you have a sketch of the PCB outline.

> Place the components freely on the PCB which will satisfy the environmental requirements, user experience, aesthetic etc.

Here is my recommended master sketch approach. Assuming you already have an PCB. Now create a sub-assembly for each component you want to use, preferably in a separate file. Assuming you are going to use simplified geometry for placing at this stage. Notice I create two construction circles as marker.
[image]

Next, go to the PCB model file. Add the same marker to a sketch. You can reuse the PCB creating sketch like I did in the screencast, or you can map a new sketch onto the PCB. Use the sketch to precisely define the component position like what will be shown in the blueprint. As shown in the screencast, you can freely move the component by moving the sketch, as long as you turn on the 'Auto update' option in sketch editor. There is an annoying problem, as the sketch will auto hide any depending object, and in this case, the top assembly. You'll have to manually toggle its visibility.
[image]

Repeat the above steps to add all the components. Finally, I'll shown you a way to export footprints after you have done the PCB assembly. The screencast only shows one component. But imagine you have placed all components. They can be exported into separate binder. You'll still need some python code convert this to footprint file, and get the relative footprint placement from the binder, though. 
[image]

> And as a side project, I'm planning to create a real time communication between FreeCAD and the PCB software (KiCAD, AeCAD, etc...) to synchronize those changes in real time.

This is an ambitious project! I actually wanted to do something similar with KiCAD, but not util the assembly WB is somewhat complete. You probably want to check out my other [project](https://github.com/realthunder/fcad_pcb).