realthunder/FreeCAD_assembly3 #56: Isn't "Create Group" supported?

ceremcem, 2018-09-04:
It seems "Create Group" is not supported. When objects are grouped they can not be moved or even their visibility can not be toggled (might be also related with #50 ). Does it have a meaning in Asm3 or is it already replaced by `Assembly Container`s?

realthunder, 2018-09-04:
There are three types of groups in FC core. The 'Create group' tree view command creates simple group, without support of coordinate space, meaning that you can't move the whole group together. The second is `App::Part` [group](https://github.com/realthunder/FreeCAD/blob/LinkStage3/src/Gui/Icons/Geofeaturegroup.svg), this is first group that supports coordinate space from upstream. It has a few restrictions on its children, which I personally think will cause problem in assembly. Therefore, I created a third type of group, the link group (accessible through tree view menu, `Link actions -> Create link group`). Assembly container is based on link group. If you intended to group objects for use as a component in an assembly, I suggest use the assembly container.

ceremcem, 2018-09-04:
Actually my first reflex is to use Assembly container but while I'm refactoring FreeCAD-library components I faced with the issue. Thanks.

closed 2018-09-04 by ceremcem