realthunder/FreeCAD_assembly3 #57: Move tool doesn't move

ceremcem, 2018-09-04:
I've downloaded a model [from FreeCAD-library](https://github.com/FreeCAD/FreeCAD-library/blob/master/Electrical%20Parts/electronic-components/headers/1x3-male-pin-header.fcstd) and started editing by adding them into Asm3 container and so on. Now the move tool doesn't seem seem to move some objects: 
[image]

## File 
[1x3-male-pin-header](https://github.com/realthunder/FreeCAD_assembly3/files/2350268/1x3-male-pin-header.fcstd.remove-extension.zip)

realthunder, 2018-09-05:
That file is designed using the old version PartDesign. You should migrate it to the new body based design first before adding to assembly. 

Before adding to assembly, simply double click any of the Pad/Pocket feature and choose Migrate.

Or, you can just used the simplified compound, pin-support-array and pin array, without all the modeling history.

ceremcem, 2018-09-05:
"Migrate"ing has solved the issue.

closed 2018-09-05 by ceremcem