realthunder/FreeCAD_assembly3 #58: Array creation problem

ceremcem, 2018-09-05:
[image]

File: [1x2-male-pin-header.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2352176/1x2-male-pin-header.fcstd.remove-extension.zip)

realthunder, 2018-09-05:
This is a bug. Will fix soon. Difference between DraftArray and DraftLinkArray is that the later is more efficient, as it does not store duplicate shape to files, but re-create the array during restore.

BTW, if you don't need pattern, then you can also make array using a normal link. Right click the object in tree view, select `Link actions -> Create link`. If you have already added a part to assembly, then use `Replace with a link`. After that, change the `ElementCount` property of the link to make an array.

closed 2018-09-05 by realthunder

ceremcem, 2018-09-05:
Array can't be recognized by Asm3 constraints: 

[image]

What would be the reason?

[case-v3.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2354596/case-v3.zip)

ceremcem, 2018-09-25:
I think this issue must be reopened since array links can not be recognized as described above.

realthunder, 2018-09-25:
I just tried the attached file. The constraint buttons is activated as expected.

ceremcem, 2018-09-26:
Am I missing something: [image]
[array-problem-20180926T1028.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2418554/array-problem-20180926T1028.zip)

commit cb847098 (2018-09-26): assembly: fix support of draft LinkArray

Fixes #58