realthunder/FreeCAD_assembly3 #59: objects in LinkGroups don't get highlighted if selected by python code

easyw, 2018-09-05:
Hi,
if you run the following code,
(where the Cone is outside a 'LinkGroup' and Cylinder is inside a 'LinkGroup')
only external objects get highlighted by the selection code.
<code>o= App.ActiveDocument.getObject('Cylinder')</code>
<code>Gui.Selection.addSelection(o)</code>
<code>print(o.Label)</code>
<code>sel = Gui.Selection.getSelection()</code>
<code>print sel[0].Label</code>
<code>o= App.ActiveDocument.getObject('Cone')</code>
<code>Gui.Selection.addSelection(o)</code>
<code>print(o.Label)</code>

[image]

realthunder, 2018-09-05:
addSelection(obj) is to select the object in the global coordinate system. And with your document the only visible objects in the global coordinate system are the link group and cone. To select the object in the link group, use something like `addSelection(LinkGroup, 'Box.')`, or refer by label, `addSelection(LinkGroup, '$Cube.')`, or refer by index `addSelection(LinkGroup,'1.')`. To select a face, `addSelection(LinkGroup, 'Box.Face1')`.

BTW, do you think it's a good idea to log those selection command to Python console? Will it be too verbose? I am already logging FC command invoking. But many commands don't work without selection.

easyw, 2018-09-06:
> addSelection(obj) is to select the object in the global coordinate system. 

But if I get an Object by its Name, it should be identified uniquely anyway... isn't it?
And in term of functionality and back compatibility it seems to work fine, at least I can get the list of selected items correctly if I use this code to add selection, but the Gui result is missing the highlight visual feedback.
Moreover, if I display the Selection View Panel, the Cylinder get listed in the selected objects.
`App.getDocument("Unnamed").getObject("Cylinder")`
And if I select it by the mouse (RMC -> 'to python console'), I get in selection view panel instead:
`Unamed.LinGroup.Cylinder.(Cylinder)`
and if I tranfert it to py console I get a different object:
`App.getDocument("Unnamed").getObject("LinkGroup")`

> BTW, do you think it's a good idea to log those selection command to Python console? Will it be too verbose? I am already logging FC command invoking. But many commands don't work without selection.

Logging to console it is always a great help in developing Macro code.

realthunder, 2018-09-06:
The highlight is not shown because the object in the global coordinate space is hidden. See the following screencast for an example. The same Box appears in three coordinate spaces. The Gui.Selection uses the second argument to determine the [selection context](../wiki/Link#user-content-subname).

[image]

easyw, 2018-09-06:
I see that with the new Link containers the power increase, but we users have to deal with the new complexity 😄 
Thx for your clarification.
M