realthunder/FreeCAD_assembly3 #61: Understanding Element Concept

ceremcem, 2018-09-05:
While reading https://github.com/realthunder/FreeCAD_assembly3/wiki/Concepts#element, it seems like that: 

Any assembly constraint MUST use two or more of points, edges or faces *directly*. In order to reduce the amount of work on broken reference issues when a major change is occurred in the model (like an already assembled component is totally replaced by another one) there is a middle layer called "Elements" that holds a map of both **our intention** (ie. "top of the screw") and the actual point/edge/face (ie. "Face017"). On a such major change, we only need to update our intentions. 

### Naming Elements 

If that's true till this point, we might consider giving names (description of our intention) to `Element`s, because (this is just a fictional example) If we wanted to replace two different sub components at once by deleting them, we can't easily update the broken `Element`s because we can't know what the intention of `Element086` was. Is that true?

realthunder, 2018-09-05:
Yes, you are right. Although, not only big changes like replacing parts, any model changes can possibly break the link. With a deep hierarchical assembly, if there is a breaking Element due to model change, you only need to update the deepest sub-assembly that directly hosting the model object. You can also check out this [tutorial](../wiki/How-to-Handle-Large-Scale-Assembly) and see how elements are manually created for different hierarchies.

ceremcem, 2018-09-06:
That's a nice concept. Thanks!

closed 2018-09-06 by ceremcem