realthunder/FreeCAD_assembly3 #62: FreeCAD Crash

ceremcem, 2018-09-05:
Again, I can't provide the reproduction steps, but here is the traceback for build:

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14975 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: e5b70fb32bd814d02dd4722de319c8d40e2913af
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
``` 

```
reeCAD 0.18, Libs: 0.18R14975 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2018
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
<asm3.sys> 98.360515 - sys_slvs.py(52): redundant constraints
<asm3.sys> 120.204103 - sys_slvs.py(52): redundant constraints
<asm3.sys> 1446.098198 - sys_slvs.py(52): redundant constraints
<asm3.sys> 43.372435 - sys_slvs.py(52): redundant constraints
<asm3.sys> 0.805653 - sys_slvs.py(52): redundant constraints
<asm3.sys> 0.708539 - sys_slvs.py(52): redundant constraints
<asm3.sys> 0.795237 - sys_slvs.py(52): redundant constraints
<asm3.sys> 0.876447 - sys_slvs.py(52): redundant constraints
<asm3.sys> 0.687543 - sys_slvs.py(52): redundant constraints
<asm3.sys> 1.189395 - sys_slvs.py(52): redundant constraints
<asm3.sys> 0.79017 - sys_slvs.py(52): redundant constraints
<asm3.sys> 325.392522 - sys_slvs.py(52): redundant constraints
<asm3.sys> 4031.043138 - sys_slvs.py(52): redundant constraints
<asm3.sys> 35.778228 - sys_slvs.py(52): redundant constraints
'Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link'Traceback (most recent call last):
  File "/home/aea/.FreeCAD/Macro/param-from-document-name.FCMacro", line 8, in <module>
    sheet.set('B3', 5)
<type 'exceptions.TypeError'>: set() argument 2 must be string, not int
'Array' cannot build shape of 'Link'Traceback (most recent call last):
  File "/home/aea/.FreeCAD/Macro/param-from-document-name.FCMacro", line 17, in <module>
    sheet.set('C3', '=%d' % x_repeat)
<type 'exceptions.TypeError'>: %d format: a number is required, not unicode
'Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link'<unknown exception traceback><type 'exceptions.SyntaxError'>: ('invalid syntax', ('/home/aea/.FreeCAD/Macro/param-from-document-name.FCMacro', 13, 12, 'regex_search = regex_pattern.search(doc_name)\n'))
'Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link'<App::Document> Document.cpp(2288): 'rpi_with_env' object 'Link' xlink property 'LinkedObject' not restored
'Array' cannot build shape of 'Link'23619.4 Selection.cpp(1425): Sub-object not found
23623 Selection.cpp(1425): Sub-object not found
'Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link''Array' cannot build shape of 'Link'Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x33060) [0x7fef92d95060]
#1  0x7fef9c58f9a2 in App::DocumentObject::getDocument() const from /opt/FreeCAD/lib/libFreeCADApp.so+0xc
#2  0x7fef9d7cb003 in Gui::ViewProviderDocumentObject::getDocument() const from /opt/FreeCAD/lib/libFreeCADGui.so+0x1f
#3  0x7fef9d5dc529 in Gui::DocumentItem::onDeleteDocument(Gui::DocumentItem*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x1c9
#4  0x7fef9d5d74a3 in Gui::TreeWidget::slotDeleteDocument(Gui::Document const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0xdf
#5  0x7fef9d5fc5d4 in boost::_mfi::mf1<void, Gui::TreeWidget, Gui::Document const&>::operator()(Gui::TreeWidget*, Gui::Document const&) const from /opt/FreeCAD/lib/libFreeCADGui.so+0x70
#6  0x7fef9d5faca0 in void boost::_bi::list2<boost::_bi::value<Gui::TreeWidget*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, Gui::TreeWidget, Gui::Document const&>, boost::_bi::rrlist1<Gui::Document const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, Gui::TreeWidget, Gui::Document const&>&, boost::_bi::rrlist1<Gui::Document const&>&, int) from /opt/FreeCAD/lib/libFreeCADGui.so+0x68
#7  0x7fef9d5f8920 in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, Gui::TreeWidget, Gui::Document const&>, boost::_bi::list2<boost::_bi::value<Gui::TreeWidget*>, boost::arg<1> > >::operator()<Gui::Document const&>(Gui::Document const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x46
#8  0x7fef9d5f6a73 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, Gui::TreeWidget, Gui::Document const&>, boost::_bi::list2<boost::_bi::value<Gui::TreeWidget*>, boost::arg<1> > >, void, Gui::Document const&>::invoke(boost::detail::function::function_buffer&, Gui::Document const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x36
#9  0x7fef9d3f0ad4 in boost::function1<void, Gui::Document const&>::operator()(Gui::Document const&) const from /opt/FreeCAD/lib/libFreeCADGui.so+0x68
#10  0x7fef9d3f0666 in boost::signals::detail::unusable boost::signals::detail::call_bound1<void>::caller<Gui::Document const&, boost::function<void (Gui::Document const&)> >::operator()<boost::signals::detail::connection_slot_pair>(boost::signals::detail::connection_slot_pair const&) const from /opt/FreeCAD/lib/libFreeCADGui.so+0x3e
#11  0x7fef9d3efa0d in boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound1<void>::caller<Gui::Document const&, boost::function<void (Gui::Document const&)> >, boost::signals::detail::named_slot_map_iterator>::dereference() const from /opt/FreeCAD/lib/libFreeCADGui.so+0x45
#12  0x7fef9d3eec96 in boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound1<void>::caller<Gui::Document const&, boost::function<void (Gui::Document const&)> >, boost::signals::detail::named_slot_map_iterator>::reference boost::iterators::iterator_core_access::dereference<boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound1<void>::caller<Gui::Document const&, boost::function<void (Gui::Document const&)> >, boost::signals::detail::named_slot_map_iterator> >(boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound1<void>::caller<Gui::Document const&, boost::function<void (Gui::Document const&)> >, boost::signals::detail::named_slot_map_iterator> const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x18
#13  0x7fef9d3ed6c8 in boost::iterators::detail::iterator_facade_base<boost::signals::detail::slot_call_iterator<boost::signals::detail::call_bound1<void>::caller<Gui::Document const&, boost::function<void (Gui::Document const&)> >, boost::signals::detail::named_slot_map_iterator>, boost::signals::detail::unusable, boost::iterators::single_pass_traversal_tag, boost::signals::detail::unusable const&, long, false, false>::operator*() const from /opt/FreeCAD/lib/libFreeCADGui.so+0x20
[... 44 lines cut ...]
#58  0x7fef9500eb5c in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xc9c
#59  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e502) [0x7fef95038502]
[... cut ...]
<asm3.sys> 5554.692338 - sys_slvs.py(52): redundant constraints
<asm3.sys> 2636.301359 - sys_slvs.py(52): redundant constraints
Rebuilding Views for: Page/Page
<asm3.sys> 8498.598847 - sys_slvs.py(52): redundant constraints
Rebuilding Views for: Page/Page
<asm3.sys> 1512.132477 - sys_slvs.py(52): redundant constraints
<asm3.sys> 56.663338 - sys_slvs.py(52): redundant constraints
<asm3.sys> 83.695015 - sys_slvs.py(52): redundant constraints
<asm3.sys> 74.74724 - sys_slvs.py(52): redundant constraints
Rebuilding Views for: Page/Page
```

realthunder, 2018-09-06:
Can you please post the file causing this crash? The crash happened when you double click a partially loaded document, and FC tries to reopen the document fully.

ceremcem, 2018-09-06:
Sure, that was in the link: https://github.com/realthunder/FreeCAD_assembly3/files/2354596/case-v3.zip

closed 2018-09-06 by realthunder

realthunder, 2018-09-06:
The drag and drop crash is fixed. But I didn't have the warning message when opening the file. Did you changed something in the file? If not, please attach your FC config files at ~/.FreeCAD/{system.cfg, user.cfg}.

ceremcem, 2018-09-06:
No, I didn't change anything. Here are the files: 
[cfg.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2355826/cfg.zip)

realthunder, 2018-09-06:
Cannot reproduce the warning. My guess is because your FC loads the documents kind of slow, and it picks up mouse over highlight event somehow during loading. I just added a fix. Please sync asm3 repo and try again.

BTW, document loading does takes extra time, whether it is partial or not. I think it is best to put the pin head models all in one files. You can have the single pin head, and various arries of pin heads all inside one file, as the pinhead library. When you use the pin head in an actual assembly, only the referred array object will be loaded, others won't even shown up in the tree view.

ceremcem, 2018-09-06:
I synced everything and there is no such warnings. Thanks. 

> You can have the single pin head, and various arries of pin heads all inside one file, as the pinhead library. 

I'll keep this in mind.