realthunder/FreeCAD_assembly3 #63: Using external geometry constraints in Sketcher in the assembly time

ceremcem, 2018-09-06:
Is it possible to use external geometry to define constraints? Like https://www.freecadweb.org/wiki/Sketcher_External, but will handle the following case: 

[image]

realthunder, 2018-09-06:
The same restriction applies, 1) the object must be in the same document, 2) the object must be in the same coordinate space as the sketch. You can use binder to achieve both.

ceremcem, 2018-09-06:
Here is a small "Projection Constraints" tutorial:

[image]

closed 2018-09-06 by ceremcem