realthunder/FreeCAD_assembly3 #64: sketcher: external reference lost when using subshapebinder

dulouie, 2018-09-06:
When using subshapebinder and sketcher external reference tool, reference can be lost or sketch may be corrupt.

[image]

[externalref_shapebinder.fcstd!](https://github.com/dulouie/screencast/raw/master/externalref_shapebinder.fcstd)

dulouie, 2018-09-08:
Use subshapebinder and export external reference from another body and edit the sketch in the reference body.
And your exported sketches are lost.
[image]

https://github.com/dulouie/screencast/blob/master/sketchexport_shapebinder.fcstd

closed 2018-09-09 by realthunder

realthunder, 2018-09-09:
The problem is fixed. I would say this is an OCCT bug, and I added a work around. The actual cause is very technical. If you are curious you can take look at the code [comment](https://github.com/realthunder/FreeCAD/blob/b8d9e29e9a7505708b1a09d616ef01c095756dae/src/Mod/Part/App/TopoShapeEx.cpp#L2087).

PS. There is a potential problem with the way you use binder. You should never bind directly to a body. Because, any new feature you add will change the body shape, and in turn the binder shape, and potentially cause loss of external geometry. You should bind to individual feature of a body, in which case, newer feature will not affect the binder.

Secondly you cannot have a sketch without any model geometry defined. It will be shown as an error, and stop further recomputation. If you only want some existing external edges, you can select those edges, and use SubShapeBinder to bind to those edges only. In fact, in your particular case, it is much easier for you to just bind to a single face.

[image]

dulouie, 2018-09-10:
I can not compile atm:

> /home/arne/src/FreeCAD/src/Mod/Part/App/TopoShape.h:187:20: error: ‘gp_Pln’ has not been declared
     bool findPlane(gp_Pln &pln, double tol=-1) const;
                    ^
In file included from /home/arne/src/FreeCAD/src/Mod/Part/App/PartFeature.h:27:0,
                 from /home/arne/src/FreeCAD/src/Mod/Part/App/FeaturePartBoolean.h:28,
                 from /home/arne/src/FreeCAD/src/Mod/Part/App/FeaturePartBoolean.cpp:32:
/home/arne/src/FreeCAD/src/Mod/Part/App/TopoShape.h:481:18: error: ‘gp_Dir’ does not name a type
            const gp_Dir &pullDirection, double angle, const gp_Pln &neutralPlane,
                  ^
/home/arne/src/FreeCAD/src/Mod/Part/App/TopoShape.h:481:61: error: ‘gp_Pln’ does not name a type
            const gp_Dir &pullDirection, double angle, const gp_Pln &neutralPlane,
                                                             ^
/home/arne/src/FreeCAD/src/Mod/Part/App/TopoShape.h:484:18: error: ‘gp_Dir’ does not name a type
            const gp_Dir &pullDirection, double angle, const gp_Pln &neutralPlane,
                  ^
/home/arne/src/FreeCAD/src/Mod/Part/App/TopoShape.h:484:61: error: ‘gp_Pln’ does not name a type
            const gp_Dir &pullDirection, double angle, const gp_Pln &neutralPlane,

And how to use the binder: It it possible to implement a routine that select the last feature of the body when you bind the body? That would prevent other people from this issue.

ceremcem, 2018-09-10:
> I can not compile atm 

I was facing with similar issues, you might want to compile everything in a virtual machine: https://github.com/ceremcem/build-freecad-asm3 Building in a virtual machine is rock-solid solution especially for daily builds.