realthunder/FreeCAD_assembly3 #66: sketcher export geometries dosen't work with KiCad StepUp

dulouie, 2018-09-06:
I want to use sketcher export geometries for pcb-edges in kicad. Is the issue related to the sketcher export tool or to stepUp from easyw?

realthunder, 2018-09-07:
@easyw Could you please look at this? I think maybe you can relax the type of Part::Part2DObject? Instead of relying on `TypeId`, you can check using `obj.isDerivedFrom('Part::Part2DObject')`

easyw, 2018-09-07:
May I have an example file and a screencast to understand the @dulouie issue better?

easyw, 2018-09-07:
ok, I see what an exported sketch is... it is not a Sketch 😄 
You can simply use the kSU button to convert it to a sketch, then Pull it to kicad 😸 
[image]

dulouie, 2018-09-08:
Ah okay..thank you that works for me:)

Another question to realthunder: Is it possible to export external reference?
Because https://github.com/FreeCAD/FreeCAD/pull/1421 is not been merged yet.

dulouie, 2018-09-08:
Nice, in my memories it dosen't work with external references, but okay now it works.
This drive me to the next bug related with shapebinder and external ref, i create a new issue for that.

closed 2018-09-11 by dulouie