realthunder/FreeCAD_assembly3 #69: Fix a point onto an object constraint is injected into the sketch without intention

ceremcem, 2018-09-07:
The "Fix a point onto an object" constraint comes out of nowhere. I've neither used this constraint, nor I know what it's for, but it pops out randomly in my sketches. It's not currently reproducable, but I come across with the case regularly. The last time was in [this](https://github.com/realthunder/FreeCAD_assembly3/issues/68#issuecomment-419341146) operation. See after clicking the Sketch under the Extrude, it pops out at the bottom vertex. (Usually it appears on the bottom vertices)

realthunder, 2018-09-07:
There is a few patches upstream fixing some problems in the sketcher. I haven't sync with upstream for quite some time. I normally do a sync before making a major release, which will hopefully be around the end of the month.

ceremcem, 2018-09-07:
Not a blocker issue, so this can wait. Thanks.