realthunder/FreeCAD_assembly3 #70: Auto naming of Elements and ElementLinks

ceremcem, 2018-09-07:
Since `Element`s are very important and should be named appropriately to remember what is what, auto naming might be improved in the following way: 

* Source entity (another element from a sub assembly or geometry from a solid) name should be used if possible. If not, a numeric postfix should be appended. 
[image]

* Constraints should refer to elements by their names for easier investigation on the model: 

[image]

realthunder, 2018-09-07:
> Source entity (another element from a sub assembly or geometry from a solid) name should be used if possible. If not, a numeric postfix should be appended.

I don't think auto naming can provide any informative name. Imaging all the Face001, Edge002, they means nothing. Besides, it requires quite some resources to track of all possible links between all the elements and their referred parts, while being extra careful to not mess with user's customized names. I prefer to give user total control of how to name their elements. If the user is confused of which element belong to which part while trying to rename it, he can use the Navigation Commands to jump forward and backward.

> Constraints should refer to elements by their names for easier investigation on the model:

This is a good idea. I have just implemented this. The label of the ElementLink and its linked Element are synchronized now. You can change either of them, and they will sync with each other. Please update both the asm3 and LinkStage3 to try this feature.

ceremcem, 2018-09-07:
Synchronization ElementLink and Element works great, thanks. :+1: 

> Imaging all the Face001, Edge002, they means nothing.

I meant drag and drop a face of a cube would create a name with `${OBJECT_LABEL}nnn` format and only for the first creation. It's very hard to synchronize for the further changes, I agree, but this is not necessary. 

This is up to you.

ceremcem, 2018-09-09:
Currently there are some issues with the synchronization: 

* Elements and ElementLinks are not synchronized if they had created before this enhancement. 
* Element and ElementLinks are not synchronized properly if there are more than one ElementLink instance for an Element: 

[image]

realthunder, 2018-09-09:
Re-label sync from ElementLink to Element is immediate, because there is one and only one Element that ElementLink links to. The sync from Element to ElementLink only happens when you perform a recomputation, because there are many ElementLink links to an Element. If you have auto assembly solve option turned on, then this two sync will be automated as well.

ceremcem, 2018-09-09:
This edge case is happened when I pulled https://github.com/realthunder/FreeCAD/commit/b8d9e29e9a7505708b1a09d616ef01c095756dae (which seems to break many elements) so if you say element links shouldn't be broken except for these extreme cases, then the case might be ignored.

bgamari, 2019-11-10:
For what is worth, I think the original suggestion of deriving the element name from the feature it is based upon is a very good one. I have found I often create a datum feature in Part Design and then immediately create an Element from it. Under this workflow I end up having to write the same name twice.

realthunder, 2019-11-11:
Use datum feature label does make sense. I'll add that in the future