realthunder/FreeCAD_assembly3 #71: FreeCAD crash

ceremcem, 2018-09-07:
# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14991 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: ca6a60371470490c74b8d7bd967bf301b4253247
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

# File 

[case-v3.bundle.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2361093/case-v3.bundle.zip)

# Reproduction
 
[image]

realthunder, 2018-09-07:
There is something wrong with PartialDocument loading logic. I'll fix it soon. Meanwhile, you can double click those greyed out document icon to manually reload the document, and this shall fix the link.

ceremcem, 2018-09-07:
Crashes are so frequent that I can't continue working. I'm falling back to `Hash: e5b70fb32bd814d02dd4722de319c8d40e2913af`: 

```console
# lxc-stop -n freecad-asm3
# cd /var/lib/lxc/freecad-asm3/
# mv rootfs/ rootfs.crashing-version
# btrfs sub snap /mnt/masa/snapshots/freecad-asm3/freecad-asm3.20180905T2029/ rootfs
Create a snapshot of '/mnt/masa/snapshots/freecad-asm3/freecad-asm3.20180905T2029/' in './rootfs'
# lxc-start -n freecad-asm3
```

closed 2018-09-10 by realthunder