realthunder/FreeCAD_assembly3 #76: Sketcher looses constraints

ceremcem, 2018-09-08:
Sketcher may loose all of its constraints, possibly after recovering from a crashed session. All constraints including external geometry might be lost. 

I'm not sure it's an upstream bug or not, I thought it would be a good idea for the rest of the users to have their guards for such a case. I'll add reproduction steps as soon as I find one.

ceremcem, 2018-09-08:
Possibly related with https://github.com/realthunder/FreeCAD_assembly3/issues/64 

There is no crash needed, 

1. Create a part with PartDesign workbench 
2. Add a sub-shape binder under the Body 
3. Create some external geometry constraints 
4. Save and close the document, 
5. Open the document 
6. External geometry constraints are lost.

ceremcem, 2018-09-13:
I can not reproduce the problem by described steps. A crash is also happened with `56a7e74336c7f7eefc540a1d13168709b5d32a09` while investigating your request and there is still no external geometry constraint loss happened.

realthunder, 2018-09-13:
Can I have the stack trace?

ceremcem, 2018-09-13:
Yes and no. I'm preparing a separate issue, will be ready in minutes.

ceremcem, 2018-09-27:
This issue is currently not reproducable but still happens. This issue is a serious one for large models. Could you please put a `help wanted` label on this?

ceremcem, 2018-09-27:
FYI and Note to myself: Last incident is caused after: 

1. I replaced an object with another object in a model (document A)
2. `A` was linked in document `B` (`B` is the main document)
3. The replaced part affected a `Cut` in `B` where the `Cut` uses `A.somepart` as a Tool and this `A.somepart` was teleported into `B` via subshape binder (`B.binder`)
4. After returning to main model (B) some subshape binders were shifted 

After fixing some subshape binders, this issue has happened.

realthunder, 2018-09-27:
Are the lost ones external constraints?

ceremcem, 2018-09-27:
*Probably*. I can only notice this case when I double click a sketch to edit where there is [a popup appears](https://user-images.githubusercontent.com/6639874/46140450-d23ea980-c259-11e8-9beb-702421616333.png)
 saying "This sketch is invalid and can not be edited". Then I open the validation tool, remove all external geometries, then sketch is opened in edit mode.

realthunder, 2018-09-27:
External geometry lost due to model change is indeed a problem stemming from upstream. I will improve this later by auto converting the lost external geometries to construction ones.

lilphil, 2018-10-08:
I've had this quite a lot, if I go back to edit a previous sketch I will find it might have lost all constraints (not just external geometry) without warning. I've noticed that if one sketch loses constraints, often other sketches near by will also lose them. The problem is, you can't tell its happened until you go back to edit something. I do not get the invalid sketch popup. I have not used links anywhere, I've not replaced an object and I don't think I've had crashes at the right time for them to be related to this.

closed 2019-01-01 by realthunder