realthunder/FreeCAD_assembly3 #78: Save All button

ceremcem, 2018-09-08:
I think this should be an upstream feature request, but if it's not that hard, it would be great to have a `Save all` button in order to easily deal with external files of a project, since we seem to deal with lots and lots of external files in Asm3.

## Rationale 

If a project (X) uses any number of external files, when the project is saved, you are asked if its dependencies should be saved too, which is okay.

If you were editing a sub-component (A) in an external file at the moment and switch to edit just another sub-component (B) and you decide to save everything, it would be possible to forget you were editing `B` so saving `B` would not ask to save `A`. You have 2 options here:

1. Remember what you are editing and if you are not editing `X`, switch to `X`, save, get back to `B`. 
2. `Close all`, reopen the project. 

Second option makes me feel safer, consumes less attention thus the brain power, so I'm sticking with the second one. 

If we had a `Save all` button, it would be very nice.

ceremcem, 2018-09-08:
The problem is that, if you have a project `X` that uses two external files, `A` and `B`, and if you are editing `A` at the moment, it won't ask you if you want to save `B` or not. You have to switch to `X` and then "save". 

Point is: Remembering the root project is hard and navigating to the root project is distracting (`X` in this case) if there are so many documents are opened.

realthunder, 2018-09-08:
Good point. I'll add that.

closed 2018-09-10 by realthunder

ceremcem, 2018-09-16:
When I click `Save all` button, following commands are dumped into the Python console: 

```
>>> App.getDocument('dc_power_jack').save()
>>> App.getDocument('rj45_90_thd').save()
>>> App.getDocument('hdmi_connector').save()
>>> App.getDocument('usb_a_type_stacked_thd').save()
>>> App.getDocument('rpi3').recompute()
>>> App.getDocument('rpi3').save()
>>> App.getDocument('rpi_with_env').save()
>>> App.getDocument('enclosure').recompute()
>>> App.getDocument('enclosure').save()
```

After that I click the close button of FreeCAD and it asks me if I want to save any changes in 'rpi-with-env' document: 

[image]

I can confirm the save order is correct according to the dependency hierarchy.

ceremcem, 2018-09-16:
Now I see the difference: In the Python console, it is instructed to save `rpi_with_env` but actual document name was `rpi-with-env`. Might that be the case?