realthunder/FreeCAD_assembly3 #80: Elements should be ordered by name

ceremcem, 2018-09-09:
If we have lots of element definitions like

* foo-middle-point
* foo-bottom-surface
* foo-front-face
* foo-bar-top-surface
* foo-baz-top-surface
* ...

it becomes hard to inspect/refactor and deduplicate* the elements if all elements are listed according to their creation sequence. If they can be ordered by their names, related elements will be listed as if they are grouped. 

> Elements may become duplicated if we create an element for a complex shape from its early stage (which is a good and solid behavior as suggested before) but if we forget that we created an edge for an object and we just click that edge and some other geometry of another object and create a constraint, then the "same element" from a "newer form of the object" is created. These should be deduplicated by human interaction at the moment.

realthunder, 2018-09-09:
I can certainly add a sorting button for this. But I think there is a misunderstanding here. The `ElementGroup` is meant to contain the interface of its owner assembly used as a `Part` for other higher level assembly.  If you strictly follow the `Element` concept described in the wiki document, then an `Element` will never be linked to the constraining `ElementLink` of the same assembly. Instead, it will always be used by some higher level assembly. Therefore, it does not make very much sense to prefix the `Element` name with the name of its associated child part. It is suppose to abstract the child part details for higher assembly.

realthunder, 2018-09-11:
Sorting function is exposed through context menu of ElementGroup.

commit c83fbce9 (2018-09-11): assembly: added a few context menu actions

* ElementGroup: sort elements by label
* Constraint: toggle disable
* Assembly: toggle freeze

Closes #80
closed 2018-09-11 by realthunder