realthunder/FreeCAD_assembly3 #81: TechDraw WB is unusable [blocker]

ceremcem, 2018-09-09:
# Description 

In TechDraw WB, clicking on a 2D object immediately opens its corresponding 3D view, which makes it impossible to edit/manipulate a 2D drawing: 

[image]

### Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14996 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: b8d9e29e9a7505708b1a09d616ef01c095756dae
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

# Working Version 

In 0.8.6, same steps produces correct results: 

[image]

realthunder, 2018-09-09:
Fixed. 

BTW, there is tree view option called `Sync view` which controls this single-click-view-jumping behavior in tree view.

closed 2018-09-09 by realthunder

ceremcem, 2018-09-10:
`Sync view` option is also nice trick, which I have it enabled (I saw the suggestion one of your wiki documents). Thanks!