realthunder/FreeCAD_assembly3 #82: Properly reporting inconsistent constraints

ceremcem, 2018-09-10:
When I upgraded to `50a8b6524f542b81473c1f94ac04a9f0dd332faf`, there is a solver exception says "inconsistent constraints": 

```
<asm3.main> 0.284546 - mover.py(186): solver exception when moving part
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/FCADLogger.py", line 83, in _catch
    return func(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 405, in solve
    return _solve(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 375, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 99, in __init__
    objName(assembly),str(e)))
RuntimeError: Failed to solve "Assembly": inconsistent constraints
```

The model was working just fine yesterday, so I'm not clear about what is going on and which are those constraints.

ceremcem, 2018-09-10:
As of `349ab4e71e8c57d9227cc202706495d636aa38f6`, the exception is not thrown for the same model, but without being sure, the following error ([screenshot](https://user-images.githubusercontent.com/6639874/45283854-a85e5680-b4e7-11e8-84b6-0b37b6e8bd2c.png)) 
 blocks recomputation: 

```
AttachEngine3D: subshape not found
```

realthunder, 2018-09-10:
The error means the sketch attached face is not found, possibly due to model change. In your case, this is due to the recent element mapping algo change. The intended algo change transition process is when you open the file, you will get a bunch or warning message about the element map version change. And you are supposed to hit the Refresh button to recompute the document, upon which the new name will be generated, and everything should be fine. But this process is currently a bit fragile, thing can go wrong if the recomputation doesn't complete in full, for various reasons. It still a working in progress.

That being said, lost of reference is going to happen for normal model modification, and you should be prepared to manually handle this type of problem from time to time. Simply re-map the sketch can fix your problem. Edit the sketch property `MapMode`. In your file, the reference in question will be inserted with a question mark (`Binder002:?Face112`) to give a clue of the originally reference.

ceremcem, 2018-09-17:
I've received same error again: 

```
<asm3.gui> 1717.60893 - gui.py(282): command "asm3CmdSolve" exception
Traceback (most recent call last):
  File "/opt/FreeCAD/Ext/freecad/asm3/FCADLogger.py", line 106, in report
    return func(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 405, in solve
    return _solve(*args,**kargs)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 375, in _solve
    Solver(assembly,reportFailed,dragPart,recompute,rollback)
  File "/opt/FreeCAD/Ext/freecad/asm3/solver.py", line 99, in __init__
    objName(assembly),str(e)))
RuntimeError: Failed to solve "Assembly": inconsistent constraints
```

# Version 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15531 (Git)
Build type: Debug
Branch: LinkStage3
Hash: d7efcd3a62d5a7da173334ef2932aef9ff8e6ad9
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

# Attachments 

[case-v3-20180917T1317.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2388145/case-v3-20180917T1317.zip)

# Reproduction

Try to `Solve constraints`, the above exception is thrown. By looking this error, I can't identify the problematic constraint. That's why this error should be improved.

realthunder, 2018-09-17:
Auto solve does not show detail of problematic constraints. They are shown only if you manually click the solve button, because finding problematic constraint takes extra time. Is it the transient error you see happen during dragging? If so, that is normal if your drag position is not solvable.

ceremcem, 2018-09-17:
Actually nothing changed in the model, except the material thickness. After fixing other constraints (Sketch constraints those depend on external geometry from sub-shape binder) I manually clicked `Solve constraints` button and this error is thrown. After clicking again (without actually moving any parts) error is gone, so I guess it's transient, yes. 

I still suggest improving the error message.