realthunder/FreeCAD_assembly3 #83: Unfolding with Sheetmetal WB fails

ceremcem, 2018-09-10:
# Description 

While it seems to be related with Sheetmetal WB, since the same file unfolds in 0.8.6 and fails to unfold in `349ab4e71e8c57d9227cc202706495d636aa38f6`, it might possibly related with LinkStage3 branch: 

### Failing (`349ab4`)

[image]

### Success (0.8.6)

[image]

# Priority 

Blocker

# Attachments 

[cut-with-binder-unfold-fails.fcstd.remove-extension.zip](https://github.com/realthunder/FreeCAD_assembly3/files/2366270/cut-with-binder-unfold-fails.fcstd.remove-extension.zip)

ceremcem, 2018-09-10:
I was building with your new commits, it's at 96%. 

> There is a visibility problem again.

I'm using this workaround though: https://github.com/realthunder/FreeCAD_assembly3/issues/50#issue-356300913

ceremcem, 2018-09-10:
Nope, still experiencing the same problem with: 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15006 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: 6a3303c1d508ec76923cce35b9e624f2d4806b66
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

I'll try to reinstall SheetMetal WB and if that doesn't help I'll perform a clean build on a clean Debian 9 installation.

ceremcem, 2018-09-10:
I've set up a clean Debian 9 machine and built LinkStage3 from scratch. Still experiencing the same problem: 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15008 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: 0aff4d7a114259ad5d82293aafbe3dc6aad771b5
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

How should we continue?

realthunder, 2018-09-11:
I reproduced problem after I updated Sheetmetal. It's fixed now.

closed 2018-09-11 by realthunder

ceremcem, 2018-09-11:
Nicely working, thanks!