realthunder/FreeCAD_assembly3 #84: Disabling auto focus on numerical inputs [upstream]

ceremcem, 2018-09-10:
Do you think disabling auto focus on numerical inputs which causes accidental data changes is hard to implement? It's clearly an upstream bug, but here are some promising searches: https://forum.freecadweb.org/viewtopic.php?f=3&t=30772

realthunder, 2018-09-10:
I think it is a much less a problem in my branch, as every property change using property editor is undoable.

ceremcem, 2018-09-10:
> every property change using property editor is undoable.

A property change is undoable *if and only if* we notice the change (while scrolling in the property window, a change might be missed)

realthunder, 2018-09-11:
I have changed property editor behavior. Please try it. If it works well, I'll submit a PR upstream.

closed 2018-09-11 by realthunder