realthunder/FreeCAD_assembly3 #85: DXF import does not work

ceremcem, 2018-09-11:
Importing DXF does not work in `63dd5552a55f2dc3ba59f1c90aa94df93f5a7c9f`:

[image]

However it works in 0.8.6: 

[image]

# Priority 

Blocker

realthunder, 2018-09-12:
I just did a merge with the upstream, and cannot reproduce this problem now. Please sync and try again.

ceremcem, 2018-09-12:
As of current build, problem still persists: 
```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15497 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: 5a4bb159157040e593e847a8b16f240eb80f954e
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

ceremcem, 2018-09-12:
That settings solved the problem: 

[image]

ceremcem, 2018-09-12:
Do you think this is a bug because they are somehow expected as enabled in a fresh install?

closed 2018-09-12 by ceremcem