realthunder/FreeCAD_assembly3 #86: Feedback system between linked documents

ceremcem, 2018-09-11:
Currently child (dependency) document has no way to get a feedback (a parameter, object, etc.) from parent (project) document and it causes cyclic dependency error. 

If we had a way to get parameters from parent documents, we could generate constraints in dependency documents which is set by the parent. 

# Usage Example

We have a PCB that should be as wide as its enclosure's inner width + 0.8mm (because PCB is attached to the enclosure by the slots on the side walls). Enclosure resides in project document and PCB is in a dependency document. We have a link to PCB document within project document.

Now, Enclosure can set its depth by looking PCB's height but PCB can't set its width by looking Enclosure width. 

To be able to solve the issue, we could perfectly merge two documents so they can use sub-shape binder to generate constraints by looking the other but this is discouraged, I guess. 

I suppose we only have to pass some parameters back to the child.

realthunder, 2018-09-12:
Are there any software to allow this kind of cyclic dependencies? I am curious to know how they handle it. You've already seen how SubShapeBinder in PartDesign handles this type of things in that modeling tutorial, i.e. it requires the user to manually sync the placement by double clicking to proceed to the next cycles. If the assembly constraints are set such that the parent body of the binder does not move after recomputation, then the cycle ends.

Anyway, for your example use case, you can create a master spreadsheet in a separate document, and refer to it in both the PCB and Enclosure file.

ceremcem, 2018-09-12:
> Are there any software that allows this kind of cyclic dependencies?

RactiveJS (a Javascript templating library) handles such cyclic dependencies, called ["two way bindings"](https://ractive.js.org/tutorials/two-way-binding/). RactiveJS even supports [recursive component usage](https://github.com/ractivejs/ractivejs.github.io/issues/123) which - I suppose - would be new to FC.

> you can create a master spreadsheet in a separate document 

That would be the way to go now, good to mention, thanks. 

--- 
Another proposal might be "Focus to this assembly" feature. In this way we won't separate the "PCB" and "Enclosure" documents, they will reside in the same document but when we need to focus (ie. freely move the parts within the PCB document), we right click the assembly container and "Focus here", which will handle the rest as if that assembly resides in a separate document.

realthunder, 2018-09-12:
> Another proposal might be "Focus to this assembly" feature.

You can achieve the same effect by simply create a link to the PCB. Actually, it is recommended to add parts only as link, and leave the original object at the root of the tree view. This can also partially solve the *Visibility* problem you've been experiencing, because now when this happens, and you select in 3D view, the tree view will sync the selection to the original object in the root of the tree for you to hide it. You can enforce this kind of 'add only as link' workflow by set the `GroupMode` property of the part group as `Auto Link` to auto create a link whenever you drag in new object.