realthunder/FreeCAD_assembly3 #87: FreeCAD Crash (linked documents are not in the tree)

ceremcem, 2018-09-12:
# What happened (not reproducable) 

1. Open a document that has linked documents 
2. See [the linked documents are not opened in the treeview](https://user-images.githubusercontent.com/6639874/45436714-1196c300-b6bc-11e8-9e62-ff811da76056.png)
3. Navigate an object that is inside a linked object through treeview
4. Try to edit (in my case, it was a Sketch)
5. FreeCAD crashes

# Version 

```
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15497 (Git)
Build type: Unknown
Branch: LinkStage3
Hash: 5a4bb159157040e593e847a8b16f240eb80f954e
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)
```

# Traceback

```
<App::Document> DocumentObject.cpp(536): Changes to partial loaded document will not be saved
<App::Document> DocumentObject.cpp(536): Changes to partial loaded document will not be saved
<App::Document> DocumentObject.cpp(536): Changes to partial loaded document will not be saved
<App::Document> DocumentObject.cpp(536): Changes to partial loaded document will not be saved
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libpthread.so.0(+0x110c0) [0x7f9e57bae0c0]
#1  /opt/FreeCAD/lib/libFreeCADBase.so(+0x1d0284) [0x7f9e5b07b284]
#2  /opt/FreeCAD/lib/libFreeCADBase.so(+0x1d0355) [0x7f9e5b07b355]
#3  /opt/FreeCAD/lib/libFreeCADBase.so(+0x1d337b) [0x7f9e5b07e37b]
#4  0x7f9e5b07e3ca in Swig_python::createSWIGPointerObj_T(char const*, void*, _object**, int) from /opt/FreeCAD/lib/libFreeCADBase.so+0x4d
#5  0x7f9e5b0172d4 in Base::InterpreterSingleton::createSWIGPointerObj(char const*, char const*, void*, int) from /opt/FreeCAD/lib/libFreeCADBase.so+0x15e
#6  0x7f9e5cc7f475 in Gui::ViewProviderPy::getRootNode() const from /opt/FreeCAD/lib/libFreeCADGui.so+0x5d
#7  0x7f9e5cc7917b in Gui::ViewProviderPy::staticCallback_getRootNode(_object*, void*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x7d
#8  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(_PyObject_GenericGetAttrWithDict+0xbb) [0x7f9e5a9ba0eb]
#9  0x7f9e5b05215a in Base::PyObjectBase::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADBase.so+0x12e
#10  0x7f9e5afe80fe in Base::BaseClassPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADBase.so+0xc8
#11  0x7f9e5b04afa8 in Base::PersistencePy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADBase.so+0xc8
#12  0x7f9e5bafe134 in App::PropertyContainerPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADApp.so+0xc8
#13  0x7f9e5ba1fbce in App::ExtensionContainerPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADApp.so+0xc8
#14  0x7f9e5cc79e02 in Gui::ViewProviderPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0xc8
#15  0x7f9e5cc6719e in Gui::ViewProviderDocumentObjectPy::_getattr(char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0xc8
#16  0x7f9e2bec4194 in PartGui::ViewProviderPartExtPy::_getattr(char const*) from /opt/FreeCAD/lib/PartGui.so+0xc8
#17  0x7f9e2aaab7f4 in PartDesignGui::ViewProviderPy::_getattr(char const*) from /opt/FreeCAD/lib/PartDesignGui.so+0xc8
#18  0x7f9e5b051ce0 in Base::PyObjectBase::__getattro(_object*, _object*) from /opt/FreeCAD/lib/libFreeCADBase.so+0x130
#19  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x488c) [0x7f9e5a9983dc]
#20  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840) [0x7f9e5a99a390]
#21  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840) [0x7f9e5a99a390]
#22  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6840) [0x7f9e5a99a390]
#23  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x86c) [0x7f9e5ab0329c]
#24  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19) [0x7f9e5a9939c9]
#25  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyRun_StringFlags+0x76) [0x7f9e5aa4b196]
#26  0x7f9e5b014cfe in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /opt/FreeCAD/lib/libFreeCADBase.so+0xd2
#27  0x7f9e5c91a8c2 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x9c
#28  0x7f9e5c91a909 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, QByteArray const&) from /opt/FreeCAD/lib/libFreeCADGui.so+0x37
#29  0x7f9e2a3ea8e8 in SketcherGui::ViewProviderSketch::setEdit(int) from /opt/FreeCAD/lib/SketcherGui.so+0x69a
#30  0x7f9e5cc471b1 in Gui::ViewProvider::startEditing(int) from /opt/FreeCAD/lib/libFreeCADGui.so+0x2d
#31  0x7f9e5cc68181 in Gu
Here is the error dump: 
i::ViewProviderDragger::startEditing(int) from /opt/FreeCAD/lib/libFreeCADGui.so+0x2f
#32  0x7f9e5c89dfb6 in Gui::Document::setEdit(Gui::ViewProvider*, int, char const*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x133a
#33  0x7f9e2a3d3de4 in SketcherGui::ViewProviderSketch::doubleClicked() from /opt/FreeCAD/lib/SketcherGui.so+0x38
#34  0x7f9e5ca68208 in Gui::TreeWidget::mouseDoubleClickEvent(QMouseEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0x176
#35  0x7f9e5444ba00 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xe10
#36  0x7f9e5480fc8e in QFrame::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e
#37  0x7f9e54934ba3 in QAbstractItemView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1e3
#38  0x7f9e5497646c in QTreeView::viewportEvent(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x3c
#39  0x7f9e5392e0a1 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x91
#40  0x7f9e543f452c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x6c
#41  0x7f9e543fcca7 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x1a47
#42  0x7f9e5c8e9a57 in Gui::GUIApplication::notify(QObject*, QEvent*) from /opt/FreeCAD/lib/libFreeCADGui.so+0xff
#43  0x7f9e5392df1d in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#44  0x7f9e543faccb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x14b
#45  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x2461a9) [0x7f9e544761a9]
#46  0x7f9e54474b5c in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xc9c
#47  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e502) [0x7f9e5449e502]
#48  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7) [0x7f9e4aec37f7]
#49  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60) [0x7f9e4aec3a60]
#50  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f9e4aec3b0c]
#51  0x7f9e5395e854 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x64
#52  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x26e5d6) [0x7f9e5449e5d6]
#53  0x7f9e5392c7ef in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#54  0x7f9e5392cb55 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#55  0x7f9e53932bd9 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#56  0x7f9e5c869c7d in Gui::Application::runApplication() from /opt/FreeCAD/lib/libFreeCADGui.so+0x22cb
#57  freecad-git(main+0xa6f) [0x558bc5c2f4f9]
#58  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1) [0x7f9e521e82e1]
#59  freecad-git(_start+0x2a) [0x558bc5c2e90a]

```

realthunder, 2018-09-13:
There is a new upstream option to set treeview to Single/Multi/Auto expand document mode. The option is exposed as main menu View->Document tree. I have changed default mode to multi document mode. I still need to investigate the crash, though.

ceremcem, 2019-04-23:
I can't reproduce the issue for a while, so I'm going to close it.

closed 2019-04-23 by ceremcem